VirtualBrainLab/Pinpoint #45: confused about stereotax vs ccf coordinates

nsteinme, 2022-03-08:
For the ccf coordinates, why not do it in "true" CCF, like measure from the 0,0,0 coordinate in the dorsal-anterior-left corner of the volume? Currently it seems like CCF is also measuring AP/ML from some assumed bregma location. I think this would be useful because it would make more clear what parts of the given coordinate correspond to some assumptions we make about skull markers.

dbirman, 2022-03-08:
Actually that's a setting as well, "Use IBL Bregma", if you turn it off it should change the coordinates to be relative to the CCF (0,0,0). I didn't think about how turning off the bregma relative coordinates would interact with the stereotaxic transform though, so I probably have to fix some issues there. So in short:

Use IBL OFF and stereotaxic OFF = standard allen CCF coordinates
Use IBL OFF and stereotaxic ON = definitely broken
Use IBL ON and stereotaxic OFF = bregma-relative CCF coordinates
Use IBL ON and stereotaxic ON = bregma-relative stereotaxic coordinates, i.e. what is in the Needles program for IBL

TODO for me:

- Fix interactions between standard allen and stereotaxic coordinates
- Separate CCF coordinate system settings into their own subpanel to make it more clear that these are a cluster of related options

nsteinme, 2022-03-08:
Sounds good. Yeah basically you should be able to implement 1) original
CCF; 2) shift to bregma but otherwise keep in CCF space; 3) full
scaling/rotation/etc to get our best guess about true stereotax

dbirman, 2022-07-26:
Fixed in v0.7.0

closed 2022-07-26 by dbirman