VirtualBrainLab/Pinpoint #53: speeding up of movement with repeated fast tapping makes things difficult

nsteinme, 2022-03-26:
There seems to be a feature which I think is intentional where the probe moves in a large jump if you have clicked one of the directional buttons a few times in succession rapidly. But what happens to me is that I'm trying to do quickly fine position and suddenly it jumps away. And I can't rapidly go back, because clicking rapidly the opposite way gets me close and then... jumps away again, now the opposite direction. There is already a modifier button for speeding up the motion. For me it'd be best if the normal tap just always did a small movement. Or you could do a hold-down-button thing where it gradually speeds up when you hold down, maybe, if you want something that doesn't require the modifier key. Or, maybe it's possible to just tweak the number of clicks needed to jump or something.

dbirman, 2022-04-07:
@dbirman 

- [ ] Single click to move a single um
- [ ] Shift + single click to move 10 um
- [ ] Click + hold or drag + hold (https://github.com/dbirman/NPTrajectoryPlanner/issues/5) to move continuously
- [ ] Shift click/hold drag/hold to move continuously x5 (or whatever the multiplier is now)

dbirman, 2022-04-12:
Bug fixed in v0.6

closed 2022-04-12 by dbirman