VirtualBrainLab/Pinpoint #100: unit testing

kjy5, 2022-07-20:
Create unit tests for things
Will require to put assembly references *everywhere* and this may not be something we want to do...

dbirman, 2022-07-28:
Added the assembly references, also refactored some of the spaghetti dependencies (see e.g. https://github.com/VirtualBrainLab/NPTrajectoryPlanner/commit/14031c95f90f70df49379e0bcaa5646398930ac9) but there will undoubtedly be more as we split the main TrajectoryPlanner folder into smaller pieces. 

It would be great if you could set up some unit tests for your code and/or for the UI elements and then do a demo for the rest of us at one of our pair coding sessions? We could do Aug 2nd/9th remote (I'll be out of town again) or in person Aug 16th.

kjy5, 2022-07-28:
Will work on this after #102 . May be easier to present what I find in-person so I will probably do that on Aug 16th

kjy5, 2022-07-28:
Helpful resources:

- https://gameconductor.net/blog/unity-test-framework-tutorial.html
- https://gameconductor.net/blog/unity-ui-test-automation-how-to-write-tests-for-your-game-interface.html

PR #118 (2022-08-04): 100 unit testing
close #100
closed 2022-08-04 by kjy5