VirtualBrainLab/Pinpoint #115: confused about coordinates again

nsteinme, 2022-08-03:
With Hailey, we were looking at the neural data from one of her recordings and trying to match it up with what we knew of her insertion coordinates, plus a photo she had of the insertion. So, we first put the tip of the probe at the known insertion coordinate and angle, then adjusted depth to go down into the brain to see what brain regions this should have gone through. However, when adjusting depth, the ap/ml coordinates didn't stay fixed, instead, they kept updating as we went deeper. I'm not sure whether this is a bug or it's a consequence of the new coordinate adjustment logic, which I know we discussed. If it's not a bug, can there be a toggle to get a mode where the numerical values of the coordinates all get adjusted independently (so that increasing depth leaves all other coords fixed and therefore drives the probe straight down its axis)?

kjy5, 2022-08-09:
I'm having trouble reproducing this. I applied a non-zero angle for each axis and moved the probe to a random location and drove down into the brain.

ap/ml/dv axes were all fixed for me when I was driving inside the brain. The online version does this as well (although there is no dv axis for the online version).

@dbirman_ would you know why this is happening?

dbirman, 2022-08-14:
@nsteinme There is no way to adjust the probe-axis depth right now (what [Z/X] used to do, whereas now they control the dorsal/ventral position). I'm implementing this in the next build (0.8) which will be up in a week or two. (closing this comment for now, tracking feature here https://github.com/VirtualBrainLab/NPTrajectoryPlanner/issues/122)

For full details: under the hood we represent all positions as the tip position (ap, ml, dv) and the probe angles (phi, theta, spin). When the probe is outside of the brain this is sufficient, when the probe is inside the brain we can calculate the insertion position on the surface and the depth relative to that, for planning surgeries. Reversing that calculation shouldn't be hard, but it is a headache because [angles are a headache ](https://www.weizmann.ac.il/sci-tea/benari/sites/sci-tea.benari/files/uploads/softwareAndLearningMaterials/quaternion-tutorial-2-0-1.pdf) and I'm still confused about it, but it works in the next version. I haven't decided how the UX will work, whether to add another set of keys for depth or have a toggle that changes the behavior of [Z/X], or only allow users to set depth through the manual entry window.

closed 2022-08-14 by dbirman