VirtualBrainLab/Pinpoint #116: Consider migrating to the new Input System

kjy5, 2022-08-04:
- Currently, project is using legacy input system which is sunsetting
- New input system also enables automated UI testing
- Both systems are currently installed and used (new input system is being used for unit testing only)

kjy5, 2022-08-11:
#117

dbirman, 2022-08-15:
This wouldn't take much work with the current setup, but before tackling this I want to separate the control functionality into its own component (ProbeController) instead of having that in ProbeManager. This would also break what @yonibrowning has developed (again, not hard to fix though). For now I'm putting this on the 1.0 milestone.

dbirman, 2023-07-19:
For controller support, see new scale controls: https://www.newscaletech.com/wp-content/uploads/MPM-Pathfinder-Xbox-Controller-Diagram.png

dbirman, 2023-07-24:
- [x] Forward axis
- [ ] Side axis
- [ ] Z/Depth
- [x] PrevProbe/NextProbe
- [ ] Key holding
- [ ] Drag detection

dbirman, 2023-07-28:
0df2db1539b13869b0569af7bd5106a49f74a11c

Not doing drag detection for now

dbirman, 2023-07-28:
Closing this issue, and breaking out remaining features into their own issues

closed 2023-07-28 by dbirman