VirtualBrainLab/Pinpoint #149: 4 shank probe zoom window

kjy5, 2022-09-02:
Not zooming in or out

dbirman, 2022-09-03:
Can't reproduce in editor -- is this possibly an interaction between sensapex and the in-plane slice code?

kjy5, 2022-09-03:
I also was unable to reproduce this while i was still in the lab today with a 1.0 probe when i was connected with a manipulator. Could just be that build?

dbirman, 2022-09-03:
Reproduced this by re-scaling the UI canvas to weird sizes, appears to be related to the button moving to a weird position and possibly a transparency issue. I removed transparency from the parent object which appears to have fixed a bug where you could click "through" to a probe behind the zoom icons. 
Also fixed a bug where clicking on the active probe would call the `SetActiveProbe` function, creating unnecessary overhead and sometimes causing probes to jump position? 

Appears to be fixed for now.

closed 2022-09-03 by dbirman