VirtualBrainLab/Pinpoint #161: 24 hour manipulator reconnection saving

kjy5, 2022-09-06:
When prompted to restore previous session, also restore manipulator state and ephys link connection when within 24 hours

dbirman, 2022-09-08:
Minor bug with this, looks like the code you pushed causes an error if the string pulled from the local prefs is empty (e.g. in my case where I've never used the ephys link). 

```
FormatException: Input string was not in a correct format.
System.Number.ThrowOverflowOrFormatException (System.Boolean overflow, System.String overflowResourceKey) (at <c2a97e0383e8404c9fc0ae19d58f57f1>:0)
System.Number.ParseInt32 (System.ReadOnlySpan`1[T] value, System.Globalization.NumberStyles styles, System.Globalization.NumberFormatInfo info) (at <c2a97e0383e8404c9fc0ae19d58f57f1>:0)
System.Int32.Parse (System.String s) (at <c2a97e0383e8404c9fc0ae19d58f57f1>:0)
System.Array.ConvertAll[TInput,TOutput] (TInput[] array, System.Converter`2[TInput,TOutput] converter) (at <c2a97e0383e8404c9fc0ae19d58f57f1>:0)
PlayerPrefs.GetRightHandedManipulatorIds () (at Assets/Scripts/TrajectoryPlanner/Utilities/PlayerPrefs.cs:308)
TrajectoryPlanner.TrajectoryPlannerManager.Start () (at Assets/Scripts/TrajectoryPlanner/TrajectoryPlannerManager.cs:150)
```

kjy5, 2022-09-08:
Oops, will address this. Is this causing a crash of some kind

commit 7ed01123 (2022-09-08): Fixed #161
PR #165 (2022-09-08): 161 24 hour manipulator reconnection saving
Fixes #161
commit 3b2b99ff (2022-09-08): 161 24 hour manipulator reconnection saving (#165)

* Fixed #161

* Use "restore previous session" wording
closed 2022-09-08 by kjy5