VirtualBrainLab/Pinpoint #178: Hosting Service Manager send error

kjy5, 2022-09-13:
```
NotSupportedException: Unexpected socket error during ping request: MessageSize
System.Net.NetworkInformation.Ping.SendPrivileged (System.Net.IPAddress address, System.Int32 timeout, System.Byte[] buffer, System.Net.NetworkInformation.PingOptions options) (at <a6f433c7914f4a8aa11562de443cf6c1>:0)
System.Net.NetworkInformation.Ping.Send (System.Net.IPAddress address, System.Int32 timeout, System.Byte[] buffer, System.Net.NetworkInformation.PingOptions options) (at <a6f433c7914f4a8aa11562de443cf6c1>:0)
System.Net.NetworkInformation.Ping.Send (System.String hostNameOrAddress, System.Int32 timeout, System.Byte[] buffer, System.Net.NetworkInformation.PingOptions options) (at <a6f433c7914f4a8aa11562de443cf6c1>:0)
System.Net.NetworkInformation.Ping.Send (System.String hostNameOrAddress, System.Int32 timeout, System.Byte[] buffer) (at <a6f433c7914f4a8aa11562de443cf6c1>:0)
System.Net.NetworkInformation.Ping.Send (System.String hostNameOrAddress, System.Int32 timeout) (at <a6f433c7914f4a8aa11562de443cf6c1>:0)
(wrapper remoting-invoke-with-check) System.Net.NetworkInformation.Ping.Send(string,int)
UnityEditor.AddressableAssets.HostingServices.HostingServicesManager.FilterValidIPAddresses (System.Collections.Generic.List`1[T] ipAddresses) (at Library/PackageCache/com.unity.addressables@1.19.19/Editor/HostingServices/HostingServicesManager.cs:550)
UnityEditor.AddressableAssets.HostingServices.HostingServicesManager.RefreshGlobalProfileVariables () (at Library/PackageCache/com.unity.addressables@1.19.19/Editor/HostingServices/HostingServicesManager.cs:421)
UnityEditor.AddressableAssets.HostingServices.HostingServicesManager.OnEnable () (at Library/PackageCache/com.unity.addressables@1.19.19/Editor/HostingServices/HostingServicesManager.cs:297)
UnityEditor.AddressableAssets.Settings.AddressableAssetSettings.OnEnable () (at Library/PackageCache/com.unity.addressables@1.19.19/Editor/Settings/AddressableAssetSettings.cs:1326)
```

dbirman, 2022-09-21:
I'm not seeing this bug -- do you get this on both mac and windows?

kjy5, 2022-09-23:
I don't remember seeing this on macOS, so I think it's just Windows

kjy5, 2022-09-27:
I will verify this later when I come in!

kjy5, 2022-09-27:
This error/warning seems to have stopped happening... not to sure when but i'll close it now

closed 2022-09-27 by kjy5