VirtualBrainLab/Pinpoint #186: Second row Probe layer view overlaps with first row

kjy5, 2022-09-15:
[image]

First row of probes display the brain layers properly, but the second row will overflow into the first layer.

dbirman, 2022-09-15:
I've never seen this before -- can you provide more info here? Is this in the editor on windows / in a build + which build target, is the canvas sized to the default size or is it in a window that has been resized?

kjy5, 2022-09-15:
- Does not happen when there are ≤ 8 probes in the scene (only happens in the expanded version when the top row has 8 probes)
- That screenshot was from the editor set to 16:9 aspect ratio
- The issue also happens in a Windows build (Mono)
- The issue occurs regardless of canvas/window size, although it is more pronounced in 16:9 (since more is seen on the vertical axis)
- In the 8-probe version, the text does overlap with the top layer, although this isn't a very serious problem: 
[image]

dbirman, 2022-10-29:
Fixed w/ 99d55dad2ea78ecf4a8630ccf7a471ef887f928a

closed 2022-10-29 by dbirman