VirtualBrainLab/Pinpoint #225: Refactor one-liner getter/setter functions into public get/set variable

kjy5, 2022-10-20:
For one-line getter/setter functions, use inline get/set variables

dbirman, 2022-11-25:
Okay I'm taking a look at this now. I disagree that serialized private fields are equivalent to public, I'm not sure why I agreed to that earlier -- I don't think exposing a field w/ `[SerializeField]` should change the naming convention. Re-iterating my argument from our earlier conversation: the naming conventions should work for *us* not the other way around, my view is that the purpose of naming conventions is to quickly see the scope of a variable. If the convention hides that information then it's a bad convention, so e.g. making a variable appear public even though it is in fact private is confusing. I can see the argument that a field exposed in the editor is in some sense public, but in that case I would say this: if the field is just getting access to some other class (e.g. the way we pass TrajectoryPlannerManager references) then it should be private to prevent someone from changing that access during runtime. If the field is settable in the editor, say a variable that controls camera min/max position or something, then consider making the field public instead of serializing a private field. Same argument for constants and static fields they need to be differentiated in some way (if you hate the SNAKE_CASE I can compromise on adding `c_` to the beginning of constants)

So I'm going to suggest the following, because I think the standard C# conventions on these is wrong:

- Use `[SerializeField] private _camelCase` for variables that need to be linked in the editor, but must not be modified by outside code at runtime
- Use `public PascalCase` for variables that need to be linked in the editor **and** can be modified by other code
- Constants should be in capitalized SNAKE_CASE
- Acronyms need to be fully capitalized (e.g. UUID, not Uuid)

dbirman, 2022-11-25:
While we're at it just to list the conventions for coordinates, later we can copy to the documentation:

- Vector3 coordinates in any `Space` should end with the suffix `Coord` (e.g. `_brainSurfaceCoord`), by default this indicates that the coordinate is in the local *transformed* space of the GameObject they are attached to
- If needed, specify whether a `Coord` is in world, space, or transformed space by appending `CoordWorld`, `CoordSpace`, or `CoordTransformed` as needed.

And then two other minor conventions that I use, but don't expect you to follow:
- When directly referring to GameObject and Transform components use the suffix `GO` and `T` (e.g. `ProbeTipGO` or `ProbeTipT`)

kjy5, 2022-11-25:
Ok, I see your reasoning for Serialized Fields. Let's do that. And I'll come back through and handle the vector naming conventions and coordinate stuff.

closed 2023-04-04 by kjy5