VirtualBrainLab/Pinpoint #229: Error- System exception-failed to load content catalog

taningh86, 2022-10-26:
Hi,
Pinpoint is not completely loading in the google chrome browser. I have no pop-ups, ads, or javascript blocked on my setting and I have also  exclusively allowed virtualbrainlab.org to use javascript. It throws the error as shown in the image below. No features seem to be working after clearing it with the clear tab. I will appreciate any suggestions to resolve this issue.
[image]

dbirman, 2022-10-29:
@taningh86 thanks for reporting this -- we're seeing this on a small number of computers and I haven't been able to re-create it myself yet. 

For now the only fix is to download the Windows desktop client from the releases page (here: https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v0.8.2), it shouldn't have this problem. Let me know if it persists in the desktop client.

dbirman, 2022-10-29:
#227

taningh86, 2022-11-01:
> @taningh86 thanks for reporting this -- we're seeing this on a small number of computers and I haven't been able to re-create it myself yet.
>
> For now the only fix is to download the Windows desktop [...]

@dbirman the desktop client worked. The interface is very intuitive to use and love the visual animation. I haven't dived deep into the tutorials yet so there must be a way it works but for some reason now the coordinates options are not active but the probe angle options are. Is it normal?

dbirman, 2022-11-01:
Yeah we're still in development right now so there will occasionally be broken features! The coordinates option is disabled because there is a UX issue with the probe positioning that I haven't solved. The longer explanation is that we display the surface coordinate and the depth of the probes, but you can technically place a probe at any location. In practice this means if you place a probe at some random location that isn't the brain surface then pinpoint will re-calculate the insertion/depth values to match the real brain surface. This is a little jarring for people if they were expecting the probe to retain the insertion coordinate they entered, and some people have asked why the program is broken and keeps changing their insertion coordinates :) -- so I need to figure out what the best system is for showing intuitively what is happening or for keeping the coordinates people enter.

dbirman, 2023-02-14:
Hi @taningh86 the issue with setting coordinates from the quick settings panel are fixed in the latest release of Pinpoint. I'm not sure whether the issues with Chrome are resolved, but for now I'm going to close this issue. I suspect it will be fixed when we update to a newer version of Unity.

closed 2023-02-14 by dbirman