VirtualBrainLab/Pinpoint #270: Change QuickSettings to removed use of FixedUpdate()

dbirman, 2023-01-03:
```
        private void FixedUpdate()
        {
            if (!_probeManager) return;
            if (_probeManager.IsEphysLinkControlled == _automaticMovementControlPanelGameObject.activeSelf) return;
            UpdateAutomaticControlPanel();
        }
```

This code probably could be changed to either depend on a UnityEvent invoked when ephys link status changes? Just an opportunity to reduce calls to Update/FixedUpdate, not urgent.

kjy5, 2023-01-04:
Oops, I thought I already fixed that. I had used that in too many places and I lost track. Will fix that tomorrow.

kjy5, 2023-01-05:
Resolved by 0b59738311c24ec5a7feb83cb1d9d1a8afa3fc0d

closed 2023-01-05 by kjy5