VirtualBrainLab/Pinpoint #274: Player Prefs gets reset or has issues loading

kjy5, 2023-01-05:
Sometime player prefs some of its setting reset which can corrupt the saved state.

**To recreate**
1. Make a scene
2. Exit play mode
3. Re-enter and click yes (everything should load fine)
4. Close out of Unity entirely
5. Re-open the project
6. Enter play mode, click yes, see that the load is corrupted

dbirman, 2023-01-05:
Leaving some notes for myself: there was a bug present in 2021.3.10 on UWP that is fixed in 3.12 https://issuetracker.unity3d.com/issues/playerprefs-dot-getfloat-returns-wrong-values-when-used-in-uwp-build, I can't tell if this is a possible source for this issue. So we may want to upgrade to the latest 2021.3.16 release if we can't find any other obvious explanation for this

kjy5, 2023-01-05:
No complaints with this!

dbirman, 2023-01-05:
Adding notes I sent on slack:

This appears to be cause by the IP address and port for ephys link getting reset to "" and 0. This could be because I don't see any HasKey checks, so if they don't exist they are getting initialized to null values? I have helper functions in the Settings class `LoadXPref` that are designed to help with this, so we could swap those out and see if that fixes the issue.

dbirman, 2023-01-06:
Added a PlayerPrefs reset option in the menu and did some testing:

- Resetting playerprefs and restarting doesn't trigger the issue (defaults are loaded correctly)
- Stopping and re-playing doesn't trigger the issue
- Having probes in the scene doesn't seem to have any effect

So if the issue still exists it's presumably triggered somehow by connecting/disconnecting w/ the Link? To-explore next time you're working.

dbirman, 2023-01-06:
Okay I'm able to re-create the bug consistently, it occurs on the Play after a compile occurs. On the subsequent plays the settings get updated

kjy5, 2023-01-06:
Ah, yeah I think it was the recompile step where issues occurred. I suppose exiting and reopening Unity has the same effect.

kjy5, 2023-01-08:
Quick update on this: I've switch the ephys link settings to use the static/property pattern now and i've added a few more checks along the way to prevent null pointer exceptions. I'm writing fixes on a branch called `patch-ephys-link-connection`. The cause for these problems is that the server settings get "forgotten" on each recompile which is why some code that was expecting the server to have it's previous configuration restored get a null pointer instead. I'm still investigating why the settings are not persisting despite them being saved.

dbirman, 2023-01-08:
Did you check whether other settings also get reset? I didn't have time to do that last week, but if it's a general issue then upgrading to the latest 2021.3 build will be worth trying.

kjy5, 2023-01-09:
I just tried again (remotely) and it appears the other settings like the Bregma one seems to be sticking. Also, the server setting were sticking too. I noticed that about 1 in 3 times I try the server settings fail to load. It's a bit confusing and I'm wondering if there is some issue with the order of operations with the data loading. All of the settings get loaded during `start`, which make sense since we need to wait for the game objects to load, but I wonder if that is somehow affecting how I launch the server connection which also happens during `start`. I thought that since my Ephys link stuff is listed later in the hierarchy than the settings script it would run later, but now I'm not 100% sure how that stuff works

dbirman, 2023-01-09:
There's no guarantee of order across objects: https://docs.unity3d.com/Manual/EventFunctions.html.

Some of the settings will fail to load properly if you put them in Awake because the downstream dependencies of their UnityEvent calls will not be fully loaded. I think it would be fine to move to Awake all settings that can be moved without causing errors, and only load things in Start if necessary for some functionality that has dependencies. Hopefully that solves your code order issues?

kjy5, 2023-01-09:
Ohhh yeah then that's it. Well my changes did fix the errors we were seeing, it's just the problem of connecting to the server. But now that makes sense because of the no guarantee on ordering of objects. I suppose I'll add a unity event to the settings load as a trigger for connecting to the server.

closed 2023-02-07 by dbirman