VirtualBrainLab/Pinpoint #285: Ephys Link setup error

BlakeRussell98, 2023-02-07:
I am trying to set up a connection between Pinpoint and a Sensapex manipulator in the lab. I can successfully start an ephys_link server, but when I try to connect to it using the Pinpoint web app (or even just open the Pinpoint web app while the server is running), I get the error "https://data.virtualbrainlab.org is not an accepted origin". Have you seen this error before and do you have any advice? Thanks.
FYI the Sensapex manipulator is connected to the PC by an ethernet to USB-c connection 
[image]

dbirman, 2023-02-08:
Yeah unfortunately this only works on standalone builds right now -- you're running into a cross-origin request issue (browser security really restricts this kind of cross-app communication). The 0.8.4 build on this page should work for you: https://github.com/VirtualBrainLab/Pinpoint/releases, if it doesn't let us know and we'll build a copy of the latest version for you to test. 

We're working on figuring out a workaround so that it works in the browser as well, we'll keep you updated.

dbirman, 2023-02-08:
@kjy5 let's discuss options here -- it's possible that the server can accept cross-origin requests actually as long as no data is being streamed back from Pinpoint?

dbirman, 2023-02-08:
Making a basic plan here:

- [ ] Build a proxy server that can echo ephys link data (make sure cors is enabled for pinpoint)
- [ ] Add a system for identifying ephys link servers -- probably on the pinpoint side this should mean copying part of a UUID instead of specifying the server address (which would always point to the proxy)
- [ ] Echoing position doesn't need security, could be accessed by just copying an ID, but we should make sure that sending commands back to ephys link is limited by some kind of user provided password

BlakeRussell98, 2023-02-09:
Hi, unfortunately I am still getting the same error using the 0.8.4 build.

dbirman, 2023-02-09:
Sorry I meant the Windows desktop build, not the web build!

dbirman, 2023-02-09:
Okay confirming here @BlakeRussell98 :

1. Using this windows build: https://github.com/VirtualBrainLab/Pinpoint/releases/download/v0.8.3/Pinpoint.zip
2. Plus running the latest ephys-link (installed in a clean venv with `pip install ephys-link --use-pep517`

I was able to get ephys link running and connected to Pinpoint. We've updated the documentation to make this more obvious, and we're hoping to push an update in the next week or two so that ephys link works through the browser.

BlakeRussell98, 2023-02-09:
I managed to get it to run and connected to Pinpoint. And when I move the Sensapex motors, the on-screen probe moves. But when I click 'enable control' in the probe UI panel, move the position of the motor (and therefore also the probe in Pinpoint), and click GO, it seems very hit or miss whether the on-screen probe and the Sensapex motor will move back to the starting position; sometimes they do not move at all and sometimes they move perfectly. Additionally, is there a way to move the target probe position in pinpoint? For example, let's say I have both the real-life Neuropixel and the pinpoint probe at Bregma, but I want to move the real-life Neuropixel to -1.9, +3.8. Can I move the target (grey) Pinpoint probe to that position, so that when I press GO, the real-life Neuropixel moves to -1.9, + 3.8?

Another thing I've noticed is that when I change the zero coordinates in the ephys link menu, they revert back to the original coordinates when I click the home button in the probe UI panel. So the probe always moves back to its original home coordinates (i.e. Bregma). I.e. I cannot change the home position of the probe.

BlakeRussell98, 2023-02-10:
Thanks for all the help and all the hard work you guys are putting in behind the scenes. I am excited to use it during surgery! If you have a demo build with some level of Pinpoint -> Sensapex 'probe control' functionality, I would be more than happy to test it out and give you any feedback that you need. Having an automated set of insertions would be amazing for my experiments so if you want me to help you out in any way my end then just let me know!

dbirman, 2023-02-10:
Sounds good, we've done one in-lab recording with the new automation so far. We're hoping to get at least one or two more recordings done to work out the user experience issues and then we'd be happy to send you a test version! We'll ping you within a few weeks

dbirman, 2023-07-14:
Hi @BlakeRussell98 just wanted to let you know that this is fully functional now, if you follow the instructions [here](https://virtualbrainlab.org/pinpoint/tutorial.html#ephys-link) you should be able to install the ephys-link Python package and then connect Pinpoint to manipulators. Both Sensapex and New Scale manipulators are working. Please let us know if you end up testing it, we'd be happy to jump on a zoom call to answer questions and help debug any issues you encounter!

BlakeRussell98, 2023-07-24:
> Hey Blake --
>
> I'm assuming here that you are up-to-date on the latest Pinpoint release, either from Github here: https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v0.9.0 or from Steam. Please double check that first!
>
> for 1, do you have the handedness set correctly for your manipulator after connecting? See: https://virtualbrainlab.org/pinpoint/tutorial.html#sensapex-handedness. If that's correct, then there could definitely be a bug, let us know.
>
> For 2, I think we should jump on a zoom call, since we don't have a tutorial right now for the copilot features. Can you email me ***@***.******@***.***> some times that work for you this week (or next) and Kenneth and I will help you debug?

Hi Dan,

I am free anytime on Wednesday if that works for you to jump on a zoom call? I appreciate your help with this.

Kind regards,
Blake

kjy5, 2023-07-28:
Hi @BlakeRussell98,
I've pushed some updates to Ephys Link which will allow it to be compatible with versions of Python 3.8 and later. The latest version of Ephys Link is 0.9.6 and can be installed with `pip install ephys-link` or `pip install ephys-link==0.9.6`. Hopefully this will make it easier for you to get the latest version on your system and let you run Pinpoint with Ephys Link!