VirtualBrainLab/Pinpoint #290: Settings for drive speed and settle timing

kjy5, 2023-02-15:


kjy5, 2023-07-18:
- Show time estimate for drive speed
- 3-speed buttons: safe, fast, testing (show warning)

dbirman, 2023-08-04:
Actually the speed settings should probably become an independent feature separated from both ProbeController and from the EphysLink code. I was thinking of changing how it works to be that if you press ctrl it slows down one speed and shift speeds up one speed (or on the controller two adjacent buttons, not sure which yet). We'd show the current speed somewhere in the viewport, and then this speed setting propagates out to all ProbeController objects.

kjy5, 2023-08-04:
[image]
UI looks something like this right now where the user would click one of these buttons to set the speed and drive.

I'm hesitant to have a universal speed used everywhere because while somebody may want to drive at a slower speed, they may want to rearrange probes in the scene faster. If it were all the same that might get confusing. (or maybe I'm misunderstanding the proposal)

Basically, I think it makes sense for there to be these speed buttons somebody could press before they drive that way they can decide what they want to do per probe

dbirman, 2023-08-04:
Hmm, good point that automation probably needs a different speed from keyboard/controller movements. Let's stick with the separated plan for now!

PR #454 (2023-08-08): 290 ephys link automation settings
close #290
closed 2023-08-08 by kjy5