VirtualBrainLab/Pinpoint #323: Add logging system

kjy5, 2023-04-04:
- [ ] Log class @dbirman 
- [ ] Create log UI @dbirman 
- [ ] Add timestamps @kjy5 
- [ ] Add automated manipulator positions (both stereotaxic and APMLDV) @kjy5

dbirman, 2023-04-06:
Add OutputLog.Log(IEnumerable<string> data)

Make sure to specify a row type as the first column of your data

kjy5, 2023-04-18:
Follow schema
```
tpmanager,timestamp(time since startup),real_date_time(datetime.now)
ephys_link,timestamp(Time.Realtime.since startup),manipulator_x,manipulator_y,manipulator_z,manipulator_d(0),phi,theta,spin,apT,mlT,dvT
```

kjy5, 2023-05-17:
The output log text box is really sluggish if you have a lot of text. We may want to think about a better way to open the log and copy text when there is a lot of it. It could be as simple as only displaying the latest x lines (since you can't even scroll on the log anyways)

dbirman, 2023-05-17:
We can change it to show the last few lines and have a button to copy out the full log. Also what's your logging rate? Probably should limit it to ~10hz or so

kjy5, 2023-05-17:
Right, the button should definitely still copy the full log. I was logging basically as fast as possible so it was stupid fast. I could limit my own Ephys logging at 10hz, but I just wanted to give the flexibility for anything else.

dbirman, 2023-05-17:
- [ ] Copy button returns the full log
- [ ] Show only latest N lines of log as text (refresh only when log updates)

kjy5, 2023-05-30:
- [ ] Rate limit to 10hz

dbirman, 2023-06-04:
Added line limit a2d5e918baee66b849f6e6a8830693d4779a1f91

closed 2023-06-04 by dbirman