VirtualBrainLab/Pinpoint #328: The Q/E keys seem to manipulate the probe in the same direction as the Z/X keys.

Aexolowski, 2023-04-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version or URL [e.g. 0.8.4 or https://data.virtualbrainlab.org/Pinpoint/]

**Additional context**
Add any other context about the problem here.

dbirman, 2023-04-12:
Hi -- the Q/E keys move the probe along the DV axis (dorsal-ventral), Z/X move the probe along the depth axis (along the probe's own axis). So when a probe is vertical, these are identical.

If the probe is off-vertical then the two axes are different, see this section of the tutorial: https://virtualbrainlab.org/pinpoint/tutorial.html#click-drag

Can you clarify if there's still a bug after looking through the tutorial explanation?

Aexolowski, 2023-04-13:
That makes total sense, thank you for the clarification!