VirtualBrainLab/Pinpoint #348: New OpenEphys format

dbirman, 2023-05-09:
Include color/acronyms:

ProbeA 0-200,STR,6ba3b9;201-383,MOs,245c4e

dbirman, 2023-06-04:
7ed9580843d2d4df6b12eaa3197a2015a51aa092

closed 2023-06-04 by dbirman

pas-calc, 2026-08-21:
The description of [OpenEphys API messages](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Probe-Viewer.html#via-config-messages) documents:
```
<probe_name>;<start1>-<end1>,<region_ID_1>,<hex_color_1>;<start2>-<end2>,...
```
But the [Pinpoint function](https://github.com/VirtualBrainLab/Pinpoint/blob/develop/Assets/Scripts/Pinpoint/Probes/ProbeManager.cs#L747-L749) does not make use of the region as a group (start-end), however writes per electrode, resulting in a very large string like
```
0,123,region,color
1,123,region,color
2,123,region,color
...
99,123,region,color
```
instead of
`0-99,123,region,color`

An option to compress that string would be to detect contiguous regions and group them.

It seems OpenEphys GUI detects it using per electrode information, though, and puts the labels only once per group.

dbirman, 2026-08-21:
https://github.com/VirtualBrainLab/Pinpoint/blob/142489d9c8213ac2aea6a76d70bbb8331a71b64f/Assets/Scripts/Pinpoint/Probes/ProbeManager.cs#L790

In any case though, this repository is not maintained anymore. We've moved to https://github.com/AllenNeuralDynamics/pinpoint, which doesn't have the OpenEphys connection right now. We'll be adding it soon.