VirtualBrainLab/Pinpoint #363: confuse about the coordinate

chongtianyifa, 2023-06-15:
Hi, 
I noticed that there is a huge difference in coordinate between Pinpoint and other software. Taking the center of VTA as an example, its AP is around 3.2 mm relative to bregma in Andy Peters’ trajectory explorer or Paxinos Atlas. In contrast, the AP is around 2.6 mm relative to bregma after MRI transformation in Pinpoint. In my experience, 3.2 mm is the right target. Do you have some explanation about the discrepancy?

Thanks,
Chris

dbirman, 2023-06-15:
Hi Chris -- that's definitely a big gap between the other atlases and Pinpoint. The difference could come from where we're defining Bregma, although at some point Andy and I coordinated on that. Do you know the Bregma coordinate in trajectory explorer or Paxinos?

This is the right place for issues, thanks.

dbirman, 2023-06-15:
One thing that may be helpful to do is to not use the MRI transform. You can turn it off in the Atlas settings (see image) by switching to "CCF Atlas". You'll find that VTA is at ~3mm from Bregma when you make that switch.

[image]

chongtianyifa, 2023-06-16:
Hi,
According to Andy's neuropixels_trajectory_explorer script, bregma was estimated at [540,44,570] in 10 um CCF atlas. I think it matches Pinpoint right now：
[image]
Hope you can figure out how to narrow the gap in the future so that we can use Pinpoint more confidently.

Thanks,
Chris

dbirman, 2023-06-16:
I just tried to quickly line up VTA in Andy's viewer and I get an AP of 3 relative to Bregma. Can you show me how you're getting an AP of 3.2? 
[image]

I also tried looking at the Allen Institute viewer to double check that there isn't an issue. In raw CCF coordinates in Pinpoint (reference coordinate set to 0,0,0) the right hemisphere center is at ap=8.35, ml=6.5, dv=5, which pretty much exactly matches what I see in the Allen Institute's viewer: http://connectivity.brain-map.org/3d-viewer?v=1&types=PLY&PLY=749 where the center is ~8.3ish, 6.5, 5. For reference here's a screenshot of the rough "center" of VTA in the allen viewer:
[image]

If you can share some more information on how you're defining VTA in each system I think we can track down where the discrepancy is coming from, but so far I'm having trouble re-creating it this. 

Also a little more context: The MRI Transform is based on a particular average of mouse brains, but if this average doesn't align well with your experience you can turn it off and plan your insertions in raw CCF space. We also have the Paxinos space that will be available soon as an option as well. You can read more about the current transform options here: https://virtualbrainlab.org/pinpoint/in_vivo_alignment.html

chongtianyifa, 2023-06-16:
I just compared the coordinate of VTA in Andy Peters’ trajectory explorer and Pinpoint more carefully. I put the coordinate of VTA center output from Pinpoint:
[image]
into Andy's trajectory explorer, 
[image]
I think it just hit the very anterior surface of VTA. I have to say that this comparison might not be fair since the transformations might be different.
According to Paxions altlas, AP of VTA ranges from -2.8 to -3.8.

dbirman, 2023-06-16:
Is it possible you're on a different version of the trajectory explorer? I downloaded the latest version today and copying your coordinates I get what's below, which is consistent with Pinpoint. Is the version of Andy's explorer you're using the latest version from the website?

[image]

As far as Paxinos, we're going to add it to Pinpoint -- hopefully in a few weeks. As far as I know the Paxinos atlas is larger than the CCF, so you can get the same effect as using the Paxinos atlas by using the bregma-lambda scaling feature in Andy's explorer or in Pinpoint and using a B-L distance > 4.1 (this feature isn't on the web right now, but it's on the development build for Pinpoint, which I can give you access to if you want to try that)

chongtianyifa, 2023-06-16:
Yes, you are right, We are using different version of trajectory explorer.
Newest:
[image]
My version:
[image]
It will be great if I can try Paxinos atlas version.

dbirman, 2023-06-16:
Great, glad we tracked that down. I'll try to send an update when we have Paxinos added, aiming for that sometime in July.