VirtualBrainLab/Pinpoint #412: Support rat atlas for the desktop version

cyf203, 2023-07-25:
I found your tool fascinating! I tried it with Qiu2018. But not sure what exactly that means. I saw you provided rat atlas download link. But I wonder if you can also add the rat atlas options in the pinpoint app. Or is there any way that I can port the local rat atlas with label and nii files to the pinpoint app? 

Also, it will be better to have a tip explanation of what the relative coordinate means. I knew each atlas have a different corrdinate system. But as a rodent researcher, I found we usually define Bregma (0,0,0) and lambda (~-9,0,0) for a rat brain. Not sure how trivial it is to calibrate this.

Thank you for the amazing work! I am looking forward to seeing more support on the rat version.

dbirman, 2023-07-25:
Thanks for checking in! There is no rat atlas in Pinpoint right now, but it is coming soon. I'm trying to build a system to make it easier to import additional atlases in the future and that's taking a little longer than just bringing the atlas in directly. It's my top priority right now so I'm hoping to have it working in the next few weeks.

So for right now we only have the mouse CCF atlas and transformations of that atlas, you can read more here https://virtualbrainlab.org/pinpoint/in_vivo_alignment.html about what the transformations like Qiu2018 do and where they came from.

cyf203, 2023-07-26:
Thank you so much! I am really looking forward to the rat atlas features. I think the current WHS rat atlas is good enough for most rat users if that's easier to integrate into the Pinpoint.

dbirman, 2023-10-19:
@cyf203 This is now complete, you can access it on the beta branch on Steam -- I can send details for how to access that if you want to test it there asap, or just wait a week or two and we'll move this onto the main release.

Also if you know of a rat skull model I would appreciate a pointer to one, thanks!

closed 2023-10-19 by dbirman