VirtualBrainLab/Pinpoint #432: Null pointer when removing probes

kjy5, 2023-07-31:
The async code is still running during a probe destroy event... need to cancel or ignore it after the probe gets destroyed

dbirman, 2023-07-31:
FYI: there's a "probe added or removed event" that triggers off of TPManager that you might want to subscribe to for this kind of thing!

kjy5, 2023-07-31:
Thanks! I'll take a look at this. I was using the Destroy and OnDestroy methods but those didn't seem to work. Do you know if those events trigger before/after Destroy happens?

dbirman, 2023-07-31:
If you're using OnDestroy there could be a garbage collector issue, OnDestroy isn't guaranteed to fire at any particular time. The ProbeManager.Destroy() function that I added does get called right before we call Destroy(probe), so that should be fine to trigger this kind of stuff

dbirman, 2023-07-31:
Ah I see -- this bug: https://github.com/VirtualBrainLab/Pinpoint/issues/429 is caused by this null pointer issue. Let me know when you push a fix so that I can push a new build out ASAP, this is on the 0.9.1 release unfortunately, which is going to cause a lot of grief if anybody tries to use it!

kjy5, 2023-07-31:
oops. Let me get onto that.

kjy5, 2023-08-01:
#433

PR #433 (2023-07-31): fix: 432 null pointer when removing probes
close #432
closed 2023-08-01 by kjy5