VirtualBrainLab/Pinpoint #437: Native collection memory leak

kjy5, 2023-08-01:
Unity throws this non-crashing error after exiting game mode every now and then:
```
A Native Collection has not been disposed, resulting in a memory leak. Enable Full StackTraces to get more details.
```

dbirman, 2023-08-02:
This is caused by Unisave because the author is using unmanaged memory and then failing to clean it up, it's not a big deal because the overhead in this particular case is very small. I think they've fixed it in newer versions, so we could probably try upgrading at some point and see if it's been resolved.

We could also just trace where this is coming from and add the .Dispose() line and then this would be cleaned up.

kjy5, 2023-08-02:
It's not really doing much so we could probably just ignore it and deal with it later.

dbirman, 2023-10-19:
Closing this, I'm going to remove Unisave in the near future and this will stop being an issue.

closed 2023-10-19 by dbirman