VirtualBrainLab/Pinpoint #478: Save zero coordinate to experiments

kjy5, 2023-08-23:
The idea is that then in subsequent runs the manipulator does not need to be zero'd at bregma and it can use the saved stereotaxic coordinate to re-align itself. This can also ensure the mouse is headfixed properly each time.

dbirman, 2023-08-24:
Hmm, not quite sure I follow this one -- I guess this is because Kim/Hailey are using one experiment per mouse basically and so this information can be stored in Pinpoint?

I think this suggests that the Accounts system should probably be overhauled to integrate with our database in some way... need to think about this more

kjy5, 2023-08-24:
They mostly liked the feature where you can move a probe back to the 0 coordinate. So basically when they set up an experiment with the same mouse on the same rig they don't have to re-zero the manipulator if they can just restore the remembered zero coordinate.

dbirman, 2023-08-24:
Hmm, so the point is that if the manipulators are in the *exact* same configuration and it's the same mouse then you can assume Bregma is in the same place and just keep running from where you left off? We could add this, but then I think we would want to store the concept of a "mouse" separately from an "experiment"

- [ ] Abstract UnisaveStorage to handle saving generic Serializable classes and handle login/logout
- [ ] Change ExperimentsManager to be some kind of DataManager class and pass Serializable struct data to UnisaveStorage
- [ ] Add a "Mouse" class that can store information such as the zero coordinate and can be saved and loaded from UnisaveStorage

kjy5, 2023-08-24:
This could be v1.1.x... what do you think?

dbirman, 2023-10-19:
Leaving a note to our future selves here: I'm going to implement a new struct `Scene` which handles this kind of data. A scene will consist of a group of insertions, the name of the active atlas and atlas transform, any rig geometry that is active, and the serialized settings string. The entire scene struct will be serialized as a JSON and we'll put/patch/get them via the vbl-dock server (with some kind of minimal password protection). We'll be able to store custom data in a scene, so this is exactly the kind of place we could save to/ load the manipulator zero offset value.

kjy5, 2025-09-05:
Will be resolved in v3.0's App UI state system

closed 2025-09-05 by kjy5