VirtualBrainLab/Pinpoint #549: Investigate issues w/ path and add better error messages for HelloSGLX

dbirman, 2023-09-15:
See messages w/ EAJ in NP Slack

emilyasterjones, 2023-09-15:
Issue is on Windows 10, occurs with v0.9.1 and v0.9.4 (haven't tested other versions) and with SpikeGLX versions 20230425 and 20230815. Connection is set up according to the tutorial but doesn't connect (hangs on `attempting connection`).

It appears to return errors in expected ways in several instances:
- if I try to connect to an older version of SpikeGLX, spikeGLX recognizes the connection but says `error [slgx_getProbeList: ERROR CmdWorker: Unknown command  [GETPROBELIST]`, as expected
- if I try to connect when SpikeGLX isn't running, `error [sglx_connect: tcpConnect: Can't connect: Connection refused (10061)]`, as expected

emilyasterjones, 2023-09-15:
Log file from a session where I attempted to connect and got the same message
[Player.log](https://github.com/VirtualBrainLab/Pinpoint/files/12623967/Player.log)

emilyasterjones, 2023-09-15:
Relevant portion of file:

> C:/Users/Midgard/Downloads/Pinpoint/Pinpoint_Data/StreamingAssets
> UnityEngine.Logger:Log(LogType, Object)
> UnityEngine.Debug:Log(Object)
> APISpikeGLX:SendAPIMessage(String, Action`1)
> APIManager:SetSpikeGLXState(Boolean)
> UnityEngine.Events.InvokableCall`1:Invoke(T1)
> UnityEngine.Events.UnityEvent`1:Invoke(T0)
> UnityEngine.UI.Toggle:Set(Boolean, Boolean)
> UnityEngine.EventSystems.ExecuteEvents:Execute(GameObject, BaseEventData, EventFunction`1)
> UnityEngine.EventSystems.StandaloneInputModule:ReleaseMouse(PointerEventData, GameObject)
> UnityEngine.EventSystems.StandaloneInputModule:ProcessMousePress(MouseButtonEventData)
> UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent(Int32)
> UnityEngine.EventSystems.StandaloneInputModule:Process()
> UnityEngine.EventSystems.EventSystem:Update()
> 
> Received probe data from SpikeGLX: ()
> UnityEngine.Logger:Log(LogType, Object)
> UnityEngine.Debug:Log(Object)
> APISpikeGLX:SetSpikeGLXProbeData(String)
> KS.UnityToolbag.<>c__DisplayClass8_0:<Invoke>b__0()
> KS.UnityToolbag.Dispatcher:Update()
> 
> ArgumentOutOfRangeException: StartIndex cannot be less than zero.
> Parameter name: startIndex
>   at System.String.Remove (System.Int32 startIndex, System.Int32 count) [0x00000] in <00000000000000000000000000000000>:0 
>   at APISpikeGLX.SetSpikeGLXProbeData (System.String allProbeDataStr) [0x00000] in <00000000000000000000000000000000>:0 
>   at KS.UnityToolbag.Dispatcher+<>c__DisplayClass8_0.<Invoke>b__0 () [0x00000] in <00000000000000000000000000000000>:0 
>   at KS.UnityToolbag.Dispatcher.Update () [0x00000] in <00000000000000000000000000000000>:0

dbirman, 2023-09-22:
Thanks for all that info, tracked it down to a blank probe data response from SpikeGLX. Working with Bill now to get this fixed, hopefully later today!

dbirman, 2023-09-22:
@emilyasterjones 

Can you confirm how you're using SpikeGLX? The **open file viewer** option is not supported by the API, you have to either launch a real acquisition or launch an existing acquisition in simulation mode.

I've clarified this in the tutorial, but it should work already with the version of Pinpoint/SpikeGLX/HelloSGLX that you have access to.

dbirman, 2023-09-22:
5a47bf8c9028466b5b3d65ceea4807dfb0ab7384

emilyasterjones, 2023-09-24:
Thanks for the tutorial update! That clarifies things.

This may be a separate issue, but the anatomy labels aren't showing up in SpikeGLX. The connection now works, but no labels appear.

[image]
On the left: simulating a previously recorded probe. Opening the probe shank view shows no anatomy.
On the right: the connection to SpikeGLX is working, and I'm able to select a probe to connect.

dbirman, 2023-09-24:
Is this just happening the first time you connect or is there just no anatomical information being sent at all? I.e. if you move the Curie_MEC probe do you then see updates in SpikeGLX?

emilyasterjones, 2023-09-24:
Sorry, I just realized this likely interacts with the other issue (probe info getting lost). The only probes that appear on the API screen (see previous screenshot) are the 5 probes whose info was deleted. I can't see the other 3 probes in this experiment, whose info is still fine. I bet there's no anatomy showing up because the info for the probe I'm connecting has been lost in Pinpoint, so there's no anatomy info to share with SpikeGLX. Is there a way to access any probes beyond the first 5 on the API screen?

dbirman, 2023-09-24:
Looks like the scrollbar on that element is inactive or missing, I'll get that fixed in the next build https://github.com/VirtualBrainLab/Pinpoint/issues/552

closed 2023-09-26 by dbirman