VirtualBrainLab/Pinpoint #551: Info for probes saved to account unexpectedly lost

emilyasterjones, 2023-09-24:
I saved 8 probes to an experiment under my account, and now the information for 5 of the probes is missing.

I created an account and saved the probes on Pinpoint online (Firefox, Windows 10), then logged into Pinpoint (Windows 10, separate machine) to connect them to SpikeGLX. On first log in, the probes had information missing, like this:
[image]
Previously, all 8 probes had unique names, colors, and coordinates. Whenever I logged into Pinpoint on this machine or on the web GUI, I get this view, with the most recent probe correct and all the rest missing information. Also, none of the colors are correct. By clicking on each probe, I was able to get all the info back (colors are still wrong). However, on recent opening, now info is missing from 5 of the probes. When I click on each probe, this is what I'm able to recover:
[image]
The names, colors, and coordinates for 4 of the probes are gone, and the coordinates for Curie_MEC are gone. Still none of the colors are correct. Between being able to recover all 8 probes and now, I have tried to connect to the SpikeGLX API several times and have restarted the computer, but haven't made any changes to the probes or my account.

dbirman, 2023-09-24:
Thanks for bringing this up -- there's a known issue that causes this, but I thought it was solved. I'll prioritize figuring out why the fix isn't working here...

dbirman, 2023-09-25:
Identified issues so far:

- [ ] On WebGL null pointer crashes when probes are disabled, possibly related https://github.com/VirtualBrainLab/Pinpoint/issues/480 
- [ ] Old datasets have incorrect ReferenceAtlas names, causing them to crash on load (and reset position?)

dbirman, 2023-09-25:
26fe86752bc1db3ddf137ff2949c05491f1d8e6b

dbirman, 2023-09-25:
- [ ] On load, the ServerProbeInsertion object appears to not get the updated data -- if a probe isn't moved before either a save event or a enable/disable event occurs the probe's data gets overridden with blank data

dbirman, 2023-09-25:
f4b5edd08b94f417a6e329069c50096d9f93b48c

dbirman, 2023-09-25:
1ed20639f03f0da66d98eb16f86c9d920bb35fb7

dbirman, 2023-09-25:
@emilyasterjones I'm pushing a build to the beta branch on Steam that may have fixed this error. Note that the current WebGL build _does not have the fix!!_

If you're using the Steam installation you can switch to the beta branch by going to the Pinpoint tab, pressing the gear icon (properties), choosing the Betas tab, and selecting "beta - Beta Releases" under the Beta Participation section. 

I'm working on recovering the original data, not sure if that will be possible yet. But if you start setting up new insertions you shouldn't see this issue anymore -- if you do see it please let me know right away and then _don't load your account again!_.

dbirman, 2023-09-25:
Fix on https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v0.9.7

closed 2023-09-25 by dbirman