VirtualBrainLab/Pinpoint #574: Gerbil atlas

Lexi13, 2023-10-24:
Would it be possible to integrate a gerbille atlas?

Here is one reference; https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5005445/

I am not sure if the authors would be able to provide the 3D information required

dbirman, 2023-10-24:
We're definitely excited to add new atlases but it would be a huge amount of work to build one directly from this image stack in the PDF. The ideal thing to have to start integrating an atlas like this is the 3D volume of annotation values. The authors who created this atlas should be able to give you that if they still have the original files (since the image stack was created from it). 

Ideally we would get from them:
- The 3D annotation volume (each voxel set to an integer corresponding it's index in the area ontology)
- The 3D reference volume (presumably an anatomical MRI/CT image)
- The ontology information (index, name, acronym, color)
- The coordinates of bregma and lambda within the volume space

Then to get the atlas into Pinpoint we would upload it to BrainGlobe (https://brainglobe.info/documentation/bg-atlasapi/adding-a-new-atlas.html) and we pull it from there. We can help with this but ideally someone else can take the lead because we have limited time! 

If you can reach out to the authors and ask about getting the 3D data volumes then we can go from there. Thanks!

Lexi13, 2023-10-24:
I have contacted the original authors to see if they can provide this information.

Thanks for the interest in helping us!

dbirman, 2024-01-28:
@Lexi13 wondering if there was any update from the authors? If not, I'm going to close this issue until we can find someone who is interested in preparing the atlas.

dbirman, 2024-01-29:
Hopefully we can find someone in the coming months who has access to the files. I have some talks coming up and will keep advertising that we need help with this (and other atlases beyond mouse/rat). Send me an email at dbirman@uw.edu any time if you find someone who can help with it. Thanks!

closed 2024-01-29 by dbirman