VirtualBrainLab/Pinpoint #577: Pinpoint require permanent Internet connection and download/upload user's data without clear concent

DenisPolygalov, 2023-10-27:
**Describe the bug**
Pinpoint require (probably persistent) internet connection for its functioning and this is not clearly stated. Moreover it is not clear that data is being downloaded and uploaded to/from user's PC. Moreover, distribution via Steam does violate IT security regulation in some research organization. Absence of clear consent for this does lead to prohibited or even illegal actions of an inexperienced user.

**To Reproduce**
Steps to reproduce the behavior:
1. Install and run Pinpoint on a PC that is not connected to Internet.
2. Pinpoint will not function (no brain pops up, no area search available, warning about absence of (or requirement of) Internet connection not shown).

dbirman, 2023-10-27:
- [ ] Look into whether atlas data can be bundled as a ZIP and loaded for local access

DenisPolygalov, 2023-10-27:
Hi Dan,

First of all let me clarify one thing from my post above – “Install and run Pinpoint on a PC” refer to the “latest build from Github: https://github.com/VirtualBrainLab/Pinpoint/releases” you mentioned.

Also, let me explain my position more clearly here. From what I understand currently Pinpoint can be obtained using 3 different methods: (1) as a web application (so it will run in user’s web browser), (2) as “standalone” binary package user can download from Github “releases” page and (3) as a “standalone” installation via Steam client.

For a reason I will mention below I choose the 2nd method and it turns out that Pinpoint require Internet connection in order to work. This is your design choice as a developer to use this method and I understand that you choose it because “it would be impractical to bundle the data for all of these into the client” as you said, but in my opinion this fact must be at least clearly stated for any user before she/he spend time dealing with Pinpoint and figure out why it does not work when no Internet connection is available. It seems like you agree with this and will add the warning and I hope will also add a statement about Internet connection requirement into documentation, main Github repository page and all future video presentations about Pinpoint.

One more thing I would like to make clear about Internet connection is that it seems to me the connection required during whole time period when Pinpoint used, so this is not just “required to download atlases when the program first loads”. The data is being downloaded on the fly, so plan like “setup Pinpoint with Internet connected, then disconnect the cable and continue to use Pinpoint offline” will not work. Am I correct about this?

The reason why I am concerned about Internet connection is obvious for anyone who ever tried to acquire high throughput low latency data streams on Microsoft Windows. As you (I hope) know - Windows is not a real time operating system, it is not controlled by user when Internet connection available. In our specific case synchronous acquisition of video stream with SpikeGLX on PC that is online lead to quite nasty outcome not even noticeable by the user who actually doing neuroscientific part of the work. Video frames will be dropped at random times causing cumulative de-synchronization of video stream and data streams acquired in parallel and other side effects.

Non-technical concerns related to permanent Internet connection required by Pinpoint is the following. Is the connection is only to download data or some information is collected from user’s PC and uploaded somewhere? If yes, you must take user’s explicit consent (“I agree” button) before allowing usage of Pinpoint. Note that such data collection might occur in the Unity engine guts, not noticeable by developer(s).

As far as I understand distribution via Steam require installation of Steam client on a user’s PC. Doing this does involve license agreement between user and Valve Corporation. For those who are not familiar according to Wikipedia “Steam is a video game digital distribution service”. I obviously don’t know about all organizations where Pinpoint used but in some of them playing video games using institution network is prohibited so some users might have fun explaining to local IT staff why Steam client is used and what kind of information it collect when running. 

Moreover, the idea behind decision to use Steam for Pinpoint distribution is that “it will automatically keep your release up to date”. Does it mean Pinpoint will be automatically updated on user’s PC without being able to opt out? If yes, a lot of “fun” might happen. The new version of Pinpoint will work (I hope), but what about compatibility with all other software and hardware components user’s rig consist of? How about downstream analysis pipeline? There is a serious reason why people reluctant to update hardware/software especially in the complicated setups like science – because updates sometimes break stuff and destroy user’s work. The reason why this happened is because software developers are only think about their software package ignoring potential problems of users who left to deal with complicated multi-component systems.

dbirman, 2023-10-28:
I think your message boils down to: "Can we please support an offline mode?"

Probably, I'll look into it.

DenisPolygalov, 2023-12-11:
Thank you, I tested the 1.0.0b release and it works OK. The alternative solution for this problem is to set up dedicated PC connected to Internet and to Neuropixels PC via direct Ethernet connection. I am considering this issue resolved for myself, so @dbirman please feel free to close it if you think your current solution is sufficient.

dbirman, 2023-12-11:
Correct, ephys-link can be run on its own computer with a local network connection to a separate computer running Pinpoint. I'll add some information about this to the documentation so that other people are aware. I think that it's not too difficult to add a fully offline version so I am still hoping to get to that.

- [ ] Add documentation about running Pinpoint offline

closed 2024-09-06 by dbirman