VirtualBrainLab/Pinpoint #579: Atlas transformations affect ephys-link position echoing

dbirman, 2023-10-31:
Got a report from a user that if you are in CCF space (the null transform) the axes are flipped, but if you are in Qiu2018 then they are correct. This was in the last release so could be fixed by now, but it sounds like somewhere the transform is either being applied or not applied incorrectly. We can debug this one together?

kjy5, 2023-10-31:
Ok, I can verify that this issue exists. Unsure why since I am not doing any computation and simply providing the raw APMLDV information from Pathfinder through `SetProbePosition`. Are the coordinate system flipped/changed between the axes somehow?

dbirman, 2023-10-31:
Yes -- SetProbePosition expects the apmldv coordinate to be in the transformed atlas space. I.e. if you are in null transform it interprets +AP to mean going -Z, but if you are in Qiu2018 transform +AP means +Z.

Your coordinates are in some Pathfinder space that points in a particular set of directions, I'm actually not sure what they are but since it works with Qiu2018 I assume that it's +AP=+Z, +ML=+X, and +DV=-Y. So to have the function work in the generic case we need to revert the axes back to world XYZ, and then match the axes to the current transform. You can do this by defining a PathfinderSpace, using the Space2World_Vector function to rotate the axes back into Unity world axes (all this will do is flip the axes), and then convert through the active atlas and active transform _Vector functions to get yourself back to the current transform space. 

Hopefully that makes sense, I can write the pseudocode out if needed!

kjy5, 2023-10-31:
Makes sense... I'm going to do this after I finish #580 which would probably mean "later"

closed 2023-11-29 by kjy5