VirtualBrainLab/Pinpoint #581: Share coordinates

RobertoDF, 2023-11-02:
HI!
Share links work only immediately after creation. What is the best way to share a set of coordinates longer term? For example this is a [link](https://data.virtualbrainlab.org/Pinpoint/?Probes=eyJBUE1MRFYiOnsieCI6MS4xMjY0MTkzMDU4MDEzOTE3LCJ5IjotMC40NTk4MjM1NDg3OTM3OTI3LCJ6IjotNC4yOTIxOTUzMjAxMjkzOTQ5fSwiQW5nbGVzIjp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjB9LCJDb29yZFNwYWNlTmFtZSI6IkNDRiIsIkNvb3JkVHJhbnNmb3JtTmFtZSI6IlFpdTIwMTgiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjAsIkNvbG9yIjp7InIiOjAuNTgwMzkyMTgxODczMzIxNSwiZyI6MC40MDM5MjE1NzQzNTQxNzE3OCwiYiI6MC43NDExNzY0ODYwMTUzMTk4LCJhIjoxLjB9LCJVVUlEIjoiYzBkOGFjY2QtNTQ0MC00ZWZlLTg0OTgtM2QwNjMzM2QxOTU4IiwiTmFtZSI6IkFDQSIsIkFQSVRhcmdldCI6IiIsIk1hbmlwdWxhdG9yVHlwZSI6IiIsIk1hbmlwdWxhdG9ySUQiOiIiLCJCcmFpblN1cmZhY2VPZmZzZXQiOjAuMCwiRHJvcDJTdXJmYWNlV2l0aERlcHRoIjp0cnVlLCJJc1JpZ2h0SGFuZGVkIjpmYWxzZX07eyJBUE1MRFYiOnsieCI6LTEuMDA5MDA3NDUzOTE4NDU3LCJ5IjotMS42NzE2MzU4NjYxNjUxNjEyLCJ6IjotNC4zNDk3MDc2MDM0NTQ1OX0sIkFuZ2xlcyI6eyJ4IjotMjA4LjAwNTA2NTkxNzk2ODc2LCJ5IjoyOS4wLCJ6IjowLjB9LCJDb29yZFNwYWNlTmFtZSI6IkNDRiIsIkNvb3JkVHJhbnNmb3JtTmFtZSI6IlFpdTIwMTgiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjAsIkNvbG9yIjp7InIiOjAuMTcyNTQ5MDI0MjI0MjgxMywiZyI6MC42Mjc0NTEwMDI1OTc4MDg4LCJiIjowLjE3MjU0OTAyNDIyNDI4MTMsImEiOjEuMH0sIlVVSUQiOiIwMWUzMWM5Yi02ODZmLTQzY2UtYmQ1NS05NzUyOGFjMWExNzciLCJOYW1lIjoiU1MiLCJBUElUYXJnZXQiOiIiLCJNYW5pcHVsYXRvclR5cGUiOiIiLCJNYW5pcHVsYXRvcklEIjoiIiwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9O3siQVBNTERWIjp7IngiOi0yLjk3MDE0MjEyNjA4MzM3NCwieSI6LTEuNTg2NDAzMDEyMjc1Njk1OSwieiI6LTQuNTU5ODA5Njg0NzUzNDE4fSwiQW5nbGVzIjp7IngiOjY3LjI4OTk5MzI4NjEzMjgxLCJ5IjozMS41MTUwMTA4MzM3NDAyMzYsInoiOjAuMH0sIkNvb3JkU3BhY2VOYW1lIjoiQ0NGIiwiQ29vcmRUcmFuc2Zvcm1OYW1lIjoiUWl1MjAxOCIsIlplcm9Db29yZE9mZnNldCI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wLCJ3IjowLjB9LCJTZWxlY3Rpb25MYXllck5hbWUiOiJkZWZhdWx0IiwiVHlwZSI6MCwiQ29sb3IiOnsiciI6MS4wLCJnIjowLjQ5ODAzOTIxNTgwMzE0NjM5LCJiIjowLjA1NDkwMTk2MTIzNzE5MjE1NywiYSI6MS4wfSwiVVVJRCI6IjQzZTA4NmQ1LWQwMjMtNDc0OS1iNjE3LTI5N2ViNTJlNGYwYiIsIk5hbWUiOiJWSVMgMi) I created yesterday that now is just empty.

dbirman, 2023-11-02:
Wow that is super mysterious. I'll start investigating...

RobertoDF, 2023-11-03:
Ok the link works! Somehow if I create an hyperlink in a Word document the link stops working. This must be related to Word though. I´ll close this.

closed 2023-11-03 by RobertoDF