VirtualBrainLab/Pinpoint #601: BestHTTP causes broken scripts

another-rishi, 2023-12-05:
I am sure this is more a user error issue related to my development environment than a bug. 

After speaking with @dbirman at SfN, it sounded reasonably straightforward to add a prefab of the Neuronexus probes and mounting adapters we use in the lab to Pinpoint. I just need to:
- load the model as an .obj file in the Assets/3D/Probes folder 
- make a prefab in Prefabs/Probes
- link it to the necessary scripts using the existing neuropixels probes as an example 
- add a RecordingSlice object 
- create a .csv with the channel map info in Assets/Data/Probes

I was hoping to do this in a fork of Pinpoint and then submit a pull request, but I've run into an issue building a local version of the project that is related to broken scripts. I gather that the EphysLink component imports the BestHTTP.SocketIO3 namespace and I need to have the BestHTTP package installed in my Unity development environment to fix the broken scripts before I can build and run the project.

To reproduce this bug, one only needs to open the project in a new development environment that doesn't have BestHTTP.

Where can I get this package? Or is there a workaround to create and debug my prefab without breaking the rest of the EphysLink code?

Thanks,
Rishi

dbirman, 2023-12-06:
Hey @another-rishi -- there are some required files that are not pushed to Github because they are paid assets. Sorry that wasn't obvious, we need to update our development documentation at some point.

You'll need to drop these files into the Assets/ folder:
https://drive.google.com/file/d/1d2YeelX-1Lj2MIBkahloaeM0QOkbCa8f/view?usp=sharing
https://drive.google.com/file/d/1KWi3aCg3yxIccZvsZ7IxmRvkPNgzmid7/view?usp=sharing
https://drive.google.com/file/d/1KhLZQamd1zZJxxSoP23vVkqYvXiR0K8-/view?usp=sharing

That should at least get you to a runnable copy of Pinpoint. Let me know how that goes? Happy to jump on zoom at some point to help debug.

Note that for the channel maps, you need to create the CSV file, but then you also need to run the Tools > Build Channel Maps function in the editor (not sure exactly what it's called). This will create an Addressable Assets file that you can then link in your ProbeManager component in your new prefab. We use Addressables for this so that the channel maps only get loaded at runtime as-needed, but it does add this extra step.

dbirman, 2023-12-06:
@another-rishi Yeah you won't be able to connect to our Unisave instance and your builds also will fail to connect without our token. It sounds like you might have made yourself a test account at Unisave (there is a free tier) to get a valid token? That would be my recommendation. I'm actually going to remove Unisave soon (next month) and replace it with a file-based save/load system with an optional "save-to-cloud" feature. So you could also just disable the Unisave code in your builds pending that change, I think to do that you just need to disable the accounts gameobject in the scene although there could be some code that depends on that which would need to be commented out.

Documentation to-dos:
- [ ] Update google drive link
- [ ] Remove information about Unisave

another-rishi, 2023-12-07:
Yes, I made a free tier Unisave account to successfully build the project.