VirtualBrainLab/Pinpoint #611: Pinpoint anatomy doesn't match on probe viewer (Open Ephys)

Andrea3v, 2023-12-13:
Hello! 
It seems I cannot get Pinpoint anatomy to align or show the correct matching areas on OpenEphys probe Viewer.

[image]

I'm running Pinpoint 1.0.2 and OpenEphys v0.6.6 with the latest ver of probe Viewer. I'm also using the CCF mouse atlas with Qiu 2018 transform (but changing transform doesn't help). 

Steps: 
I'm zeroing the probe at Bregma in the setup and in Pinpoint, then I connect to Sensapex'a manipulator using Ephys-link. I the move to the insertion point and I'm connecting to OpenEphys using the API tab to send the anatomy info. I then use the offset in the Ephys Link tab to align the anatomy of pinpoint to the brain surface and drive the probe to target.

thank you in advance for any help or suggestion you may have
Andrea

dbirman, 2023-12-14:
Thanks for reporting this, I think there's a rotation issue somewhere. Should be an easy fix but I won't be able to push it out until I'm back from vacation in January.

In the meantime, if you're on Steam we have a beta branch that is tagged to an older version (v0.9) which should work properly. You can also download older Windows releases from Github directly, e.g. this one: https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v0.9.7

If you're using the web application you can also see all the available versions by going to https://data.virtualbrainlab.org/. Pinpoint 0.9.8 or 0.9.7 should work for you (hopefully).

Andrea3v, 2023-12-14:
Hi Dan! thank you for your reply. indeed it seems a rotation issue. I'm running v0.9.7 and the anatomy now matches between pinpoint and open-ephys. but to obtain the same probe orientation in space I now have to replace the yaw-pitch values from 90-95 to 270-5. I noticed that was also causing the axis to be inverted in the beta version in the slice view vs. the 3D volume. 
hope this helps getting it fixed...
happy vacation!
~Andrea

dbirman, 2023-12-15:
I think you're seeing the effect of two changes we made going from 0.9 -> 1.0:

We changed how we define pitch so that instead of zero meaning vertical in 1.0 zero now means angle off the horizontal plane, which is more consistent with how most people measure that angle.

The yaw change could be because in 0.9 the "Use IBL conventions" setting is turned on by default (in probe settings), that was a bug, if you go and turn it off I think the yaw should then match what you had in 1.0?

dbirman, 2024-01-18:
Partial fix on 1.1.2 d9005396e023259414958cc6cb98bd45b437bfe8

dbirman, 2024-01-18:
Fixed, pending 1.1.1 f42fe54d566043bd632991727fa15833b35596ca

closed 2024-01-18 by dbirman