VirtualBrainLab/Pinpoint #613: Pinpoint anatomy doesn't match on probe viewer (SpikeGLX)

Virginia9733, 2023-12-14:
**Describe the bug**
Similar to the issue described here: https://github.com/VirtualBrainLab/Pinpoint/issues/611
It seems like PinPont's anatomy doesn't match the anatomy shown in the SpikeGLX IMRO table.

We updated the newest version of PinPoint.

Using Neuropixels 1.0: 
Anatomy in PinPoint looks like this:
[image]
But in SpikeGLX look like this:
[image]

Using Neuropixels 2.0: 
Anatomy in PinPoint looks like this:
[image]
But in SpikeGLX look like this:
[image]

Anatomy in PinPoint looks like this:
[image]
But in SpikeGLX look like this:
[image]

Looking forward to hearing from you.

dbirman, 2023-12-14:
Thanks this is a known issue, we'll have it fixed in early January.

In the meantime, if you're on Steam we have a beta branch that is tagged to an older version (v0.9) which should work properly. You can also download older Windows releases from Github directly, e.g. this one: https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v0.9.7

#611 
#602

dbirman, 2024-01-18:
Partial fix on 1.1.2 d9005396e023259414958cc6cb98bd45b437bfe8

dbirman, 2024-01-18:
Fixed, pending 1.1.1 f42fe54d566043bd632991727fa15833b35596ca

closed 2024-01-18 by dbirman