VirtualBrainLab/Pinpoint #618: rotation changed and is confusing/broken now?

nsteinme, 2024-01-08:
When I click and drag to rotate the brain, something is different about the way it used to be, such that now I cannot rotate the brain into all possible angles. The view starts looking top-down, and no matter how I click-drag, I cannot achieve a sagittal view with the AP axis going left-right (I can get sagittal only if the nose is pointing up) - I think this used to be possible. If I double-click the blue axis to snap to sagittal, now I can no longer get top-down with the nose pointing up, only top-down with nose pointing left. The upshot of all this is that if I want a particular off-axis viewing angle, it's really hard for me to figure out how to achieve it - it seems like I first have to figure out which axis to snap to, then execute some unnatural-seeming series of turns. Not sure whether I'm describing this clearly - can hop on zoom some time if helpful.

dbirman, 2024-01-08:
Yeah this is a bug that came from trying to fix the rotations in Urchin so that it would be more logical what the outcome is if you set them from code, which worked, but then unexpectedly caused the mouse-controlled rotations to constantly cause gimbal lock (which is why it's hard to get to a sagittal view without knowing some secret jedi tricks).

On the to-do list!

nsteinme, 2024-01-08:
Maybe one option could be to have alt-click-drag, shift-, and ctrl- be set
to each lock two of the three axes, and the dragging only sets the third
axis. This would always allow sort of "manual" setting of the angle.

dbirman, 2024-03-30:
Partial fix, a760abdf922d43d3113fe5976999cbe415656e5c

- [ ] Add locked axis controls?

jonahpearl, 2024-10-24:
Apologies if I'm missing something, but the "hold spacebar" to access the third rotation mode doesn't seem to be working, at least for me. It works sometimes, ie if I click the blue axis on the view widget, then it works for a bit. But when the brain is at certain angles, ie if I click the red axis, hitting spacebar doesn't affect how the brain rotates. Thanks!

dbirman, 2024-10-29:
This is related to the issue above. The current controls have "gimbal lock" where if you rotate the brain in some directions you lose an axis. I hope to fix it eventually but haven't had time