VirtualBrainLab/Pinpoint #635: Coordinates nonzero at Bregma

Virginia9733, 2024-01-31:
**Describe the bug**
We updated the latest version of Pinpoint but found that the coordinates look a bit peculiar.
The bregma was not set to zero and it's a bit less intuitive to match the probes to the position that was described in the atlas coordinates.

Also, the brain regions look a bit different between PinPoint and SpikeGLX.
In PinPoint, all recording channels evenly distributed on the bottom of four shanks were all in dorsal CA1, while in SpileGLX, the depth of recording regions was a bit higher, and some of the recording channels were not in CA1.

[image]
[image]
[image]

dbirman, 2024-01-31:
Hey @Virginia9733 thanks for reporting these issues, some of these may already be fixed. Can you confirm which version of Pinpoint you are on? The mis-aligned channels bug shouldn't be a problem anymore in version 1.1.1.

The Bregma coordinate not getting set to the default value is a bug, I'll go see what's going on there.

dbirman, 2024-01-31:
Thanks, I'm investigating... will try to get this patched soon.

dbirman, 2024-02-02:
Bregma coordinate default fix on 1.1.2 a515adf6e5a1bffd3433ad9e253b711d9b190ae5

dbirman, 2024-02-02:
- [ ] Mis-aligned channels fix

closed 2025-01-09 by dbirman