VirtualBrainLab/Pinpoint #637: Yaw and roll do the same thing?

chris-angeloni, 2024-01-31:
**Describe the bug**
Yaw and roll rotate NP2.4 probes along the same axis. I need to adjust roll and it isn't working.

**To Reproduce**
1. Add NP.0 (4 shank) probe.
2. Hit "1" or "3" to change yaw. (or type angle in yaw)
3. Hit "," or "." to change roll. (or type angle in roll)

**Screenshots**
[image]
[image]

**Log**
[pinpoint_log.txt](https://github.com/VirtualBrainLab/Pinpoint/files/14115188/pinpoint_log.txt)

chris-angeloni, 2024-01-31:
To make this a bit more clear, my desired insertion angle is this, which I think should be yaw 0, pitch 90, and roll 23, unless I am thinking about this all wrong: 
[image]

chris-angeloni, 2024-01-31:
Hmm, I was able to get it by playing around with the rotations quite a bit, but it is not very intuitive!
[image]

closed 2024-01-31 by chris-angeloni

dbirman, 2024-02-01:
Sorry I didn't see this before you figured it out -- I find it helps a lot to think about yaw/pitch/roll in order, since that's how they are implemented internally. So first yaw your probe to the angle your manipulator will be at, then pitch it to the angle you will attach it to the manipulator, and finally roll it to get the probe surface pointing in the direction you want.

So in this case, you want the manipulator to be mounted facing to the left, so Yaw = -90, then you want the probe to be tilted at off vertical at 23 deg, so Pitch = +67 (90-23), and finally you want the probe facing posterior so you set Roll = +90.

[image]

I have an alternate angle system that will come in a future release that lets you define angles as angles off the sagittal or coronal plane (or around AP/ML if you prefer). For this particular recording that scheme would be easier, since this is just a probe that is 23 degrees off sagittal, plus a roll.