VirtualBrainLab/Pinpoint #652: Add custom Rig to render

SpencerBowles, 2024-02-16:
Hello, our lab is trying to use Pinpoint to plan some multi-probe Neuropixel experiments. We are using custom 3d printed headplates to create insertion points for the probes. We would like to be able to import the model of our headplate into Pinpoint in order to determine where we need our holes to be in the final prints. 

Is it possible to add custom items to the Rig section of Pinpoint? If so, what formatting is needed for the files to be imported smoothly?

Thanks!

dbirman, 2024-02-16:
Hey Spencer, thanks for checking on this. I've had it on my to-do list for a while to make this easier and it is partially functional, but I keep delaying it to work on other things. I'll bump it up the to-do list and try to get it done later this month!

I am planning two ways of doing this:

Option one is that you will (soon) be able to make a pull request against another repository called "pinpoint-rig-models" or something, and any files in that folder will get automatically added to the next build of Pinpoint. These files will then be available to all users of Pinpoint, and because they are built-in to the program won't require any extra steps to access.
Option two will be for files you want to keep private, you will be able to upload 3D models to your current instance of Pinpoint through a Python API, specifying the position of the model's origin coordinate and any translation/rotation.

For now -- if you have a .obj 3D model file you want added you can just send it to me by email (dbirman@uw.edu) or drop it here and I can add it in. Please include a square image to use as the example image and let me know if the origin of the object should be positioned at a particular location. The current rig objects are placed relative to bregma (i.e. the (0,0,0) origin coordinate in the 3D model is intended to be at Bregma).

SpencerBowles, 2024-02-23:
Ok awesome! I will send you the files. What do you mean by a square image?

chris-angeloni, 2025-05-15:
Hi, just checking if there has been movement on this issue? I would also like to optimize multi probe placement using custom chronic implant models to avoid collisions during surgery!

dbirman, 2025-05-15:
Custom rig objects are possible now through Urchin but it's tricky to use (and there's no tutorial). We're going to make a big push for our full 2.0 release before July this year and this is one of the things that should get a lot clearer.

Same thing for probes, I have on my roadmap to streamline how probes get ingested and make it possible to more easily add custom probes through the probe-library. These would end up in the next build. Adding custom probes at runtime is tricky, I'll think more about whether this is possible to support.