VirtualBrainLab/Pinpoint #655: add Kim atlas

WeissShahaf, 2024-02-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Please add Kim atlas, important for brainstem implants such as PAG.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

https://data.mendeley.com/datasets/2svx788ddf/1
also implamented in Brainglobe: https://brainglobe.info/documentation/bg-atlasapi/usage/atlas-details.html

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thanks,
Shahaf

dbirman, 2024-02-19:
Hi Shahaf -- the BrainGlobe implementation of the Kim atlas has a bug that prevents it from being used in Pinpoint. I'm working with the people who uploaded it to try to fix the issue.

Thanks!

WeissShahaf, 2024-03-26:
any news on fixing the issue to get Kim atlas on Pinpoint?

dbirman, 2024-03-26:
The issue is fixed on the BrainGlobe side, but they haven't pushed it as an updated package yet. As soon as it is, I'll run the pipeline on my side and we should be good to go.

StempelLab, 2024-03-29:
we are using Brainglobe for registering the probes. is there a bug we should know about or is it only an issue of formatting to get the data into pinpoint?

dbirman, 2024-03-29:
@StempelLab Is this a question related to the Kim atlas, or something else?

StempelLab, 2024-03-29:
> @StempelLab Is this a question related to the Kim atlas, or something else?

the Kim atlas

dbirman, 2025-01-09:
We're now pulling all available atlases from BrainGlobe-atlasapi -- if the atlas is working properly and is in the standard format it should show up in Pinpoint in a future build.

closed 2025-01-09 by dbirman