VirtualBrainLab/Pinpoint #659: info copy from web UI doesn't make sense

WeissShahaf, 2024-02-27:
**Describe the bug**
A clear and concise description of what the bug is.
as i lower the probe in Pinpoint, more site become void, instead of more sites being in the brain.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'help'
2. Click on 'API'
3. click copy channel data
4. pate in notepad++

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log**
After re-creating the bug in your build of Pinpoint, please find and attach the Player.log file. This has important debugging information that we need to fix issues.

On **Windows** the Player.log file is located at: %USERPROFILE%\AppData\LocalLow\VirtualBrainLab\Pinpoint\Player.log
On **Linux**: ~/.config/unity3d/VirtualBrainLab/Pinpoint/Player.log
In a **browser** the Player.log file is written to the console. Please copy and paste the entire contents of the console into your issue, or into a file attached to your issue. To access the console: 

    In Firefox, press Ctrl-Shift-K on Windows or Command-Option-K on a Mac.
    In Chrome, press Ctrl-Shift-J on Windows or Command-Option-J on a Mac.
    In Safari, go to Preferences > Advanced > Develop, and press Command-Option-C.
    In Microsoft Edge or Internet Explorer, press F12.

dbirman, 2024-02-27:
@WeissShahaf Can you tell me the version number that you are using? There was recently a bug that caused the API values to be mis-matched to the probes, but I thought I already fixed it. You can see the version number by pressing ESC and looking in the Help tab, it's listed on the bottom of the menu.

WeissShahaf, 2024-02-28:
i'm using the browser / web version V1.1.1 Unity 2022.3.18f1

dbirman, 2024-02-28:
Thanks, I'm investigating!