VirtualBrainLab/Pinpoint #667: RAT issues on pinpint 1.0 and 1.1

Andrea3v, 2024-03-15:
Hi. 
I noticed the following bugs while using different versions of pinpoint and rat atlases:
 
1- on 1.1 and ephys-link-Beta none of the atlases are loading

2- on v1.0 (Beta included) atlases are loading but there's a mismatch between the 3D areas and the areas listed on the probe/slice view:
[image]
this also translates into errors while sending the info to OpenEphys using API:
[image]
(the areas here are again different from both the listed ones and the 3D, maybe a problem with rotation?).

3- on v1.0 the Skull (rig feature) has a z-offset:
[image]

thank you for looking into this,
~Andre

dbirman, 2024-03-16:
Hey @Andrea3v -- what platform are you on? I assume Windows? If so, can you please launch Pinpoint v1.1 (which is not loading the atlas for you), allow it to load as far as it will go, and then send me this file: %USERPROFILE%\AppData\LocalLow\VirtualBrainLab\Pinpoint\Player.log

This would really help us debug that particular issue.

The skull rig is a mouse skull, not a rat skull -- if you have a rat skull 3D model please send it to me and I will add it into Pinpoint!

The other issue with v1.0 is resolved in 1.1 but obviously not helpful until we get that working for you.

Andrea3v, 2024-03-16:
Hi @dbirman 

I'm running Windows10 with pinpoint from Steam. 
in v1.1 I get as far as Pinpoint is asking to reset the scene and if I want to proceed: if I hit 'Yes' nothing happens. I've attached the player.log file

[Player.log](https://github.com/VirtualBrainLab/Pinpoint/files/14621955/Player.log)

I also tried on a different computer where I'm running Windows 11 and pinpoint 1.1.2 (from GitHub, in case it was something with Steam?). same issue (Player2). [Player2.log](https://github.com/VirtualBrainLab/Pinpoint/files/14622425/Player2.log).

will send a rat skull STL shortly (: 

thank you!
~Andre

dbirman, 2024-03-16:
Got it -- I see the error and I'll have it fixed in the next build. Hopefully that will resolve the alignment issues you were seeing as well.