VirtualBrainLab/Pinpoint #682: request: guide to adding new probes

StempelLab, 2024-04-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
a walkthrough / guide on adding new probes to Pinpoint - for local use.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
a jupyeter notebook / video on how to make and add custom probes to pinpoint
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
i haven't found anything online that describes it other than users attempting on their own
**Additional context**
Add any other context or screenshots about the feature request here.

dbirman, 2024-04-26:
It's been on my to-do list to get this working for a while now, thanks for the reminder!

WeissShahaf, 2024-09-25:
@dbirman any news on a probe template guide?

dbirman, 2024-09-26:
I've made a bit of progress on this, you can see the plan here: https://github.com/VirtualBrainLab/probe-library

I'll be putting more time toward this before SfN as part of our Pinpoint v2 push. Sorry it's so delayed.