VirtualBrainLab/Pinpoint #684: uMp-3-NP support

kjy5, 2024-05-02:
I am not sure asking this here is appropriate but is uMp-3-NP supported by PinPoint Software? We tried to use it with the latest version v1.1.6 but it did nork. Please let me know if there is another version to download. I apologize if this is not appropriate.

_Originally posted by @20180220 in https://github.com/VirtualBrainLab/Pinpoint/issues/458#issuecomment-2089981943_

20180220, 2024-05-07:
Thank you very much for your reply.
Well, we simply tried to use Pinpoint softwware v1.1.6. with  a uMp-3-NP Manipulator of Multiprobe rig (Sensapex),
but we could not connect the software with it. We were able to do so with uMp-4. 
[image]
The messages we got in the software were different, so I attached it for FYI. 
We really apprecaite your support.

Best Regards,

kjy5, 2024-05-07:
Ok cool, thanks for the update and the error message. Assuming the image on the left is with a uMp-3, it looks like they did successfully connect but is running into trouble with further API calls. I will take a look at this and get back to you soon.

20180220, 2024-05-07:
yes, that is correct. The left one is the one which did not work. 
Thank you very much for your support! 
We really apprecaite it.

Best Regards,

kjy5, 2024-05-09:
Hi, I've made a new patch release of Pinpoint that hopefully addresses this issue you are seeing. Please let me know if it works for you! Thank you!

https://github.com/VirtualBrainLab/Pinpoint/releases/tag/v1.1.7

20180220, 2024-05-09:
Hi, Thank you very muc for releasing the new version. 
We will try and let you know if it work!
Again, thank you!

20180220, 2024-05-10:
Hi, I have jsut tried it, and it worked!! Thank you very much for your support! 
We truely apprecaite it.

Best Regards,

kjy5, 2024-05-10:
Sweet! So glad to hear this build fixed your issues. I'll close this issue for now but feel free to message back if you have any questions!

closed 2024-05-10 by kjy5

20180220, 2024-05-10:
Sure, I will. Again, thank you so much!