VirtualBrainLab/Pinpoint #690: question:two dimensions do not align well

Batter-Wang, 2024-05-10:
[image]
hi~May I have a question? I found that slices of two dimensions do not align well. I also check the sixth edition of Paxios&Watson the rat brain in stereotaxic coordinates, and there were same problem. How can I find the exact location of the electrode? Is  slice dyeing the only way ?

Batter-Wang, 2024-05-10:
[image]
[image]

dbirman, 2024-05-17:
Sorry for the delay I didn't see this issue right away.

If the Paxinos atlas also has this issue it might be a real problem with the atlas. I suspect though that it's just a bug and that the sagittal slice is inverted or offset in some weird way. 

To your other question -- the *only* way to know the true position of the probe is histology. Neither the Paxinos or the Waxholm atlas are accurate estimates of the true position of your probe. Although the atlases can help with initial targeting I highly recommend doing histology afterwards to identify the real trajectory of your probe!