VirtualBrainLab/Pinpoint #738: np2.0 quad 3d model

nsteinme, 2024-09-18:
Hi Dan, I got ahold of the NP 2.0 Quad models - probe and headstage at these links: [probe](https://www.dropbox.com/scl/fi/cjqs4g4mugpkiaidqp52z/NPM_FLEX_02-03-014-004.STEP?rlkey=qyd2ysa4w1vrqxxoy9l5k13l3&dl=0), [hs](https://www.dropbox.com/scl/fi/hshpqtpsy455m76bv8bbj/NPM_HS_32-05-025-002.STEP?rlkey=j59y41arjbsbmvem4wfdvofu6&dl=0) (though I guess the HS is just for your interest since I'm not sure there is a point of doing anything with it). Hope the ".step" file is a useful format. Could you add this probe as an option for the 'add new probe' menu? Thanks!

dbirman, 2024-09-18:
While I'm adding this I might as well add the ultra probe -- it's a 2.0 base right, but do you have the channel map configuration info somewhere?

nsteinme, 2024-09-18:
No, Ultra is a 1.0 probe. Here are the x (
https://www.dropbox.com/scl/fi/3828z05b1vg2r9u39zcmw/channels.xcoords.npy?rlkey=b4usowmkulyt2ljnyyugh9c4a&dl=0)
and y (
https://www.dropbox.com/scl/fi/vbzx5l7l8k9xujp6v7iaf/channels.ycoords.npy?rlkey=m3ymwlid5t5gjcfm7xttslsh1&dl=0)
coordinates of the sites (it's just a 48x8 rectangular grid with 5x5 um
sites and 1 um gaps for 6 um center-to-center pitch). Those coordinates
relative to the bottom corner of the grid - it would be another ~200 um to
the tip as in a normal 1.0 probe.

dbirman, 2024-09-18:
Perfect, thanks

nsteinme, 2025-08-19:
Hi @dbirman we need this now - is it something @kjy5 can help with? Thanks!!

kjy5, 2025-08-21:
Yeah I used some CAD program like FreeCAD

dbirman, 2025-08-22:
@nsteinme I'm going to set up the new pinpoint probe library to pull from the spikeinterface probeinterface_library package. Do you know if the new probe you want is already in this list? https://github.com/SpikeInterface/probeinterface_library/tree/main/imec I'm not sure where these 4-digit numbers come from do you?

nsteinme, 2025-08-22:
I don't think it is - looks like we could ask Jennifer - I also don't know
exactly what all the numbers mean. but I don't think those files have the
geometry of the probe base right? Shanks of Quad are going to be the same
as 4-shank 2.0 as far as I understand (so NP2014 should work) but what we
need this for is to make sure we can find geometries where the bases don't
collide. The 3d files I added here have this.

kjy5, 2025-08-22:
We have the geometries sorted out through the files you sent! I think the concern was whether this new probe has a new electrode layout or things like that we needed to map. This would be the information that these data files would tell us.

dbirman, 2025-08-22:
Yeah expanding on what Kenneth said, I want to pull the channel/shank geometry from this library if possible, headstage/base geometry will come in separately. So the hope is that all probe channel geometry that we might have to deal with are in there. But yeah sounds like we can just make a PR and add it as another option if it's missing based off of the 2014 layout.