VirtualBrainLab/Pinpoint #752: craniotomy coordinates aren't subtracting bregma properly

nsteinme, 2024-10-22:
It looks like the range of coordinates that you are allowed to choose from is set to go around bregma (e.g. ML goes -5700 to 5700) but it is actually going around the CCF origin or somewhere, see below. The craniotomy with the given coordinates is the one that visually looks like it's at bregma. So one consequence is, I cannot get to the right hemisphere because the ML coordinate maxes at 5700 which is right on the midline. I just have the default MouseCCF atlas selected. Thanks - 

[image]

Here's a state link: https://data.virtualbrainlab.org/Pinpoint/?Probes=eyJBUE1MRFYiOnsieCI6NS45MDA5MDE3OTQ0MzM1OTQsInkiOjQuOTg3NTI4ODAwOTY0MzU1OSwieiI6Mi44OTk1Njk1MTE0MTM1NzR9LCJBbmdsZXMiOnsieCI6MTM1LjAsInkiOjYwLjAsInoiOi03NS4wfSwiQXRsYXNTcGFjZU5hbWUiOiJhbGxlbl9tb3VzZV8yNXVtIiwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoiIiwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjI0LCJDb2xvciI6eyJyIjowLjEyMTU2ODYyNzY1NTUwNjE0LCJnIjowLjQ2NjY2NjY2ODY1MzQ4ODE4LCJiIjowLjcwNTg4MjM3MDQ3MTk1NDQsImEiOjEuMH0sIlVVSUQiOiIyYzZlOWQwNS0yNjcwLTQ0NzMtYjNlZi1jZTE1MDVjZGJjOWMiLCJOYW1lIjoiY2hwbC0yYzZlOWQwNSIsIkFQSVRhcmdldCI6IiIsIk51bUF4ZXMiOjAsIk1hbmlwdWxhdG9ySUQiOiIiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiRGltZW5zaW9ucyI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9O3siQVBNTERWIjp7IngiOjUuODkxMzk1MDkyMDEwNDk4LCJ5Ijo0LjAxODgyOTM0NTcwMzEyNSwieiI6My40NTMyMzE1NzMxMDQ4NTg2fSwiQW5nbGVzIjp7IngiOjkwLjAsInkiOjYwLjAsInoiOi03NS4wfSwiQXRsYXNTcGFjZU5hbWUiOiJhbGxlbl9tb3VzZV8yNXVtIiwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoiIiwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjI0LCJDb2xvciI6eyJyIjowLjgzOTIxNTY5NTg1ODAwMTcsImciOjAuMTUyOTQxMTgyMjU1NzQ0OTQsImIiOjAuMTU2ODYyNzUwNjQ5NDUyMiwiYSI6MS4wfSwiVVVJRCI6ImI4M2EwMDZkLWRlMjEtNGU4My1iZDkwLTBiMzRkYWU2ZmRjMyIsIk5hbWUiOiJDUC1iODNhMDA2ZCIsIkFQSVRhcmdldCI6IiIsIk51bUF4ZXMiOjAsIk1hbmlwdWxhdG9ySUQiOiIiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiRGltZW5zaW9ucyI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9O3siQVBNTERWIjp7IngiOjYuNDMzMzExNDYyNDAyMzQ0LCJ5Ijo0Ljg4MDM5NTg4OTI4MjIyNywieiI6NC4wOTgwNzYzNDM1MzYzNzd9LCJBbmdsZXMiOnsieCI6NDUuMCwieSI6NjAuMCwieiI6LTYwLjB9LCJBdGxhc1NwYWNlTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0iLCJBdGxhc1RyYW5zZm9ybU5hbWUiOiIiLCJTZWxlY3Rpb25MYXllck5hbWUiOiJkZWZhdWx0IiwiVHlwZSI6MjQsIkNvbG9yIjp7InIiOjAuNTgwMzkyMTgxODczMzIxNSwiZyI6MC40MDM5MjE1NzQzNTQxNzE3OCwiYiI6MC43NDExNzY0ODYwMTUzMTk4LCJhIjoxLjB9LCJVVUlEIjoiZmExNTllNDYtYzk2Yy00ODZlLWI0YzItZWQ3NDFhYzEwMzIwIiwiTmFtZSI6IkNMLWZhMTU5ZTQ2IiwiQVBJVGFyZ2V0IjoiIiwiTnVtQXhlcyI6MCwiTWFuaXB1bGF0b3JJRCI6IiIsIlplcm9Db29yZE9mZnNldCI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wLCJ3IjowLjB9LCJEaW1lbnNpb25zIjp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjB9LCJCcmFpblN1cmZhY2VPZmZzZXQiOjAuMCwiRHJvcDJTdXJmYWNlV2l0aERlcHRoIjp0cnVlLCJJc1JpZ2h0SGFuZGVkIjpmYWxzZX07eyJBUE1MRFYiOnsieCI6OC43NTM0OTYxNzAwNDM5NDYsInkiOjMuMzc0MzU5MzY5Mjc3OTU0LCJ6IjoxLjU4OTczMTIxNjQzMDY2NH0sIkFuZ2xlcyI6eyJ4IjowLjAsInkiOjYwLjAsInoiOi03NS4wfSwiQXRsYXNTcGFjZU5hbWUiOiJhbGxlbl9tb3VzZV8yNXVtIiwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoiIiwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjI0LCJDb2xvciI6eyJyIjowLjk2ODYyNzQ1Mjg1MDM0MTgsImciOjAuNzEzNzI1NTA3MjU5MzY4OSwiYiI6MC44MjM1Mjk0MjIyODMxNzI2LCJhIjoxLjB9LCJVVUlEIjoiNGUxNjY1N2UtNDg4Yi00YmFmLWExOTktMDhkNGUxNTU5NTI3IiwiTmFtZSI6IlZJU3AtNGUxNjY1N2UiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfTt7IkFQTUxEViI6eyJ4Ijo3Ljg0MzczMDkyNjUxMzY3MiwieSI6Ni4wNTI3NjEwNzc4ODA4NTksInoiOjEuNjMxNzY1MjQ2MzkxMjk2NH0sIkFuZ2xlcyI6eyJ4IjotNDUuMCwieSI6NjAuMCwieiI6LTc1LjB9LCJBdGxhc1NwYWNlTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0iLCJBdGxhc1RyYW5zZm9ybU5hbWUiOiIiLCJTZWxlY3Rpb25MYXllck5hbWUiOiJkZWZhdWx0IiwiVHlwZSI6MjQsIkNvbG9yIjp7InIiOjAuMDkwMTk2MDgwNTA1ODQ3OTMsImciOjAuNzQ1MDk4MDU0NDA5MDI3MSwiYiI6MC44MTE3NjQ3MTcxMDIwNTA4LCJhIjoxLjB9LCJVVUlEIjoiZjVjOTgyZDktYzIzZi00NjYyLTg1ZTMtYTRlYzdkNWQyNzI0IiwiTmFtZSI6IlJTUHYtZjVjOTgyZDkiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfTt7IkFQTUxEViI6eyJ4IjozLjk2MTIyMzM2Mzg3NjM0MjksInkiOjYuMDU1MjQ3MzA2ODIzNzMwOSwieiI6My4xNjQyODM3NTI0NDE0MDY0fSwiQW5nbGVzIjp7IngiOi0xMzUuMCwieSI6NjAuMCwieiI6LTc1LjB9LCJBdGxhc1NwYWNlTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0iLCJBdGxhc1RyYW5zZm9ybU5hbWUiOiIiLCJTZWxlY3Rpb25MYXllck5hbWUiOiJkZWZhdWx0IiwiVHlwZSI6MjQsIkNvbG9yIjp7InIiOjAuNTQ5MDE5NjM0NzIzNjYzMywiZyI6MC4zMzcyNTQ5MTE2NjExNDgwOSwiYiI6MC4yOTQxMTc2NTkzMzAzNjgwNiwiYSI6MS4wfSwiVVVJRCI6IjVjOTEwYWRmLWZhZTAtNGRjZi05NGZjLTA3NGQ4ZDYzYTY2OCIsIk5hbWUiOiJQTC01YzkxMGFkZiIsIkFQSVRhcmdldCI6IiIsIk51bUF4ZXMiOjAsIk1hbmlwdWxhdG9ySUQiOiIiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiRGltZW5zaW9ucyI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9O3siQVBNTERWIjp7IngiOjMuNzMyMDQ5NDY1MTc5NDQzNSwieSI6NC4zOTk5OTk2MTg1MzAyNzMsInoiOjIuNDY0MTAyMDI5ODAwNDE1fSwiQW5nbGVzIjp7IngiOjE4MC4wLCJ5Ijo2MC4wLCJ6IjotOTAuMH0sIkF0bGFzU3BhY2VOYW1lIjoiYWxsZW5fbW91c2VfMjV1bSIsIkF0bGFzVHJhbnNmb3JtTmFtZSI6IiIsIlNlbGVjdGlvbkxheWVyTmFtZSI6ImRlZmF1bHQiLCJUeXBlIjoyNCwiQ29sb3IiOnsiciI6MS4wLCJnIjowLjU5NjA3ODQ1NTQ0ODE1MDYsImIiOjAuNTg4MjM1MzE4NjYwNzM2MSwiYSI6MS4wfSwiVVVJRCI6IjBmZDRlODdlLWU4NjMtNDBjNS05M2EwLTRlNDRhZDQyNmI1OSIsIk5hbWUiOiJNT3MtMGZkNGU4N2UiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfTt7IkFQTUxEViI6eyJ4Ijo2LjgxMDUzMjA5MzA0ODA5NiwieSI6Ny41NjM4MjA4Mzg5MjgyMjMsInoiOjMuMjEyMjczMTIwODgwMTI3fSwiQW5nbGVzIjp7IngiOi05MC4wLCJ5Ijo2MC4wLCJ6IjotNzUuMH0sIkF0bGFzU3BhY2VOYW1lIjoiYWxsZW5fbW91c2VfMjV1bSIsIkF0bGFzVHJhbnNmb3JtTmFtZSI6IiIsIlNlbGVjdGlvbkxheWVyTmFtZSI6ImRlZmF1bHQiLCJUeXBlIjoyNCwiQ29sb3IiOnsiciI6MC44NTg4MjM1Mzc4MjY1MzgxLCJnIjowLjg1ODgyMzUzNzgyNjUzODEsImIiOjAuNTUyOTQxMjAzMTE3MzcwNiwiYSI6MS4wfSwiVVVJRCI6ImY0YzZjOGZiLWZlNzctNDc1Ny05MmYwLWFkNmRiMGJjOTM1NiIsIk5hbWUiOiJDQTMtZjRjNmM4ZmIiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfQ==&Settings=eyJTaG93U3VyZmFjZUNvb3JkIjp0cnVlLCJTaG93QnJlZ21hQXhpcyI6dHJ1ZSwiU2hvd0luUGxhbmVTbGljZSI6dHJ1ZSwiR2hvc3RJbmFjdGl2ZVByb2JlcyI6dHJ1ZSwiR2hvc3RJbmFjdGl2ZUFyZWFzIjpmYWxzZSwiVW5pdHNJblVNIjp0cnVlLCJTaG93QWxsUHJvYmVQYW5lbHMiOmZhbHNlLCJQcm9iZVBhbmVsSGVpZ2h0IjoxNDQwLjAsIkRldGVjdENvbGxpc2lvbnMiOnRydWUsIlJvdGF0ZUFQTUwyUHJvYmVBeGlzIjp0cnVlLCJQcm9iZVNwZWVkIjoyLCJTaG93QXRsYXMzRFNsaWNlcyI6MCwiUmVsYXRpdmVDb29yZCI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoibnVsbCIsIkJyZWdtYUxhbWJkYVJhdGlvIjoxLjAsIkVwaHlzTGlua1NlcnZlcklQIjoiIiwiRXBoeXNMaW5rU2VydmVyUG9ydCI6ODA4MSwiRXBoeXNMaW5rUmlnaHRIYW5kZWRNYW5pcHVsYXRvcnMiOiIiLCJBY2NvdW50c0xvZ2luVG9nZ2xlIjp0cnVlLCJPcGVuRXBoeXNBUElUb2dnbGUiOmZhbHNlLCJPcGVuRXBoeXNBUElUYXJnZXQiOiJodHRwOi8vbG9jYWxob3N0OjM3NDk3IiwiU3Bpa2VHTFhBUElUb2dnbGUiOmZhbHNlLCJTcGlrZUdMWEFQSVRhcmdldCI6IjEyNy4wLjAuMTo0MTQyIiwiU3Bpa2VHTFhIZWxsb1BhdGgiOiJDOlxcSGVsbG9TR0xYLXdpblxcSGVsbG9TR0xYLmV4ZSIsIkFQSVVwZGF0ZVJhdGUiOjEwLjAsIlVzZUlCTEFuZ2xlcyI6ZmFsc2UsIkF4aXNDb250cm9sIjp0cnVlLCJVc2VBY3JvbnltcyI6dHJ1ZSwiVXNlQmVyeWwiOnRydWUsIkNhbWVyYVpvb20iOjUuMCwiQ2FtZXJhUm90YXRpb24iOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkF0bGFzTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0ifQ==

dbirman, 2024-10-22:
Ugh that's annoying... not sure where this came from. On the to-do list

kjy5, 2024-11-11:
Just to make sure, @nsteinme, did you intentionally set the reference coordinate to something other than Bregma? Based on the picture, the origin point looks like it is in the top left of your screen when it typically would be at Bregma.

nsteinme, 2024-11-12:
I did not intentionally set the reference coordinate, no. Not urgent to solve this or #753 so no need for any quick patch or whatever.

kjy5, 2024-11-13:
Ok yeah in that case this particular situation can be resolved by moving the reference coordinate back to Bregma so you can access the desired coordinates (settings > atlas > reference coordinate > Bregma). It's still a bug though as mentioned in #753 but it'll be fixed once the data system is online.