VirtualBrainLab/Pinpoint #753: loading a link doesn't come up with the same thing?

nsteinme, 2024-11-06:
I generated [this link](https://data.virtualbrainlab.org/Pinpoint/?Probes=eyJBUE1MRFYiOnsieCI6NC41MDkyMjAxMjMyOTEwMTYsInkiOjMuOTE4MDg5Mzg5ODAxMDI1NiwieiI6My45MDQzMTgzMzI2NzIxMTl9LCJBbmdsZXMiOnsieCI6MTM1LjAsInkiOjYwLjAsInoiOi03NS4wfSwiQXRsYXNTcGFjZU5hbWUiOiJhbGxlbl9tb3VzZV8yNXVtIiwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoiIiwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjI0LCJDb2xvciI6eyJyIjowLjEyMTU2ODYyNzY1NTUwNjE0LCJnIjowLjQ2NjY2NjY2ODY1MzQ4ODE4LCJiIjowLjcwNTg4MjM3MDQ3MTk1NDQsImEiOjEuMH0sIlVVSUQiOiIyYzZlOWQwNS0yNjcwLTQ0NzMtYjNlZi1jZTE1MDVjZGJjOWMiLCJOYW1lIjoiUEwtMmM2ZTlkMDUiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfTt7IkFQTUxEViI6eyJ4Ijo2LjAsInkiOjMuMzY2MDI1NDQ3ODQ1NDU5LCJ6IjozLjgzMDEyNzcxNjA2NDQ1M30sIkFuZ2xlcyI6eyJ4Ijo5MC4wLCJ5Ijo2MC4wLCJ6IjotNzUuMH0sIkF0bGFzU3BhY2VOYW1lIjoiYWxsZW5fbW91c2VfMjV1bSIsIkF0bGFzVHJhbnNmb3JtTmFtZSI6IiIsIlNlbGVjdGlvbkxheWVyTmFtZSI6ImRlZmF1bHQiLCJUeXBlIjoyNCwiQ29sb3IiOnsiciI6MC44MzkyMTU2OTU4NTgwMDE3LCJnIjowLjE1Mjk0MTE4MjI1NTc0NDk0LCJiIjowLjE1Njg2Mjc1MDY0OTQ1MjIsImEiOjEuMH0sIlVVSUQiOiJiODNhMDA2ZC1kZTIxLTRlODMtYmQ5MC0wYjM0ZGFlNmZkYzMiLCJOYW1lIjoiQ1AtYjgzYTAwNmQiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfTt7IkFQTUxEViI6eyJ4Ijo2LjQzMzMxMTQ2MjQwMjM0NCwieSI6NC44ODAzOTU4ODkyODIyMjcsInoiOjQuMDk4MDc2MzQzNTM2Mzc3fSwiQW5nbGVzIjp7IngiOjQ1LjAsInkiOjYwLjAsInoiOi02MC4wfSwiQXRsYXNTcGFjZU5hbWUiOiJhbGxlbl9tb3VzZV8yNXVtIiwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoiIiwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjI0LCJDb2xvciI6eyJyIjowLjU4MDM5MjE4MTg3MzMyMTUsImciOjAuNDAzOTIxNTc0MzU0MTcxNzgsImIiOjAuNzQxMTc2NDg2MDE1MzE5OCwiYSI6MS4wfSwiVVVJRCI6ImZhMTU5ZTQ2LWM5NmMtNDg2ZS1iNGMyLWVkNzQxYWMxMDMyMCIsIk5hbWUiOiJDTC1mYTE1OWU0NiIsIkFQSVRhcmdldCI6IiIsIk51bUF4ZXMiOjAsIk1hbmlwdWxhdG9ySUQiOiIiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiRGltZW5zaW9ucyI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9O3siQVBNTERWIjp7IngiOjkuMjk0MjI4NTUzNzcxOTczLCJ5IjozLjAsInoiOjEuOTAxOTIzNjU2NDYzNjIzfSwiQW5nbGVzIjp7IngiOjAuMCwieSI6NjAuMCwieiI6LTc1LjB9LCJBdGxhc1NwYWNlTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0iLCJBdGxhc1RyYW5zZm9ybU5hbWUiOiIiLCJTZWxlY3Rpb25MYXllck5hbWUiOiJkZWZhdWx0IiwiVHlwZSI6MjQsIkNvbG9yIjp7InIiOjAuOTY4NjI3NDUyODUwMzQxOCwiZyI6MC43MTM3MjU1MDcyNTkzNjg5LCJiIjowLjgyMzUyOTQyMjI4MzE3MjYsImEiOjEuMH0sIlVVSUQiOiI0ZTE2NjU3ZS00ODhiLTRiYWYtYTE5OS0wOGQ0ZTE1NTk1MjciLCJOYW1lIjoiVklTcC00ZTE2NjU3ZSIsIkFQSVRhcmdldCI6IiIsIk51bUF4ZXMiOjAsIk1hbmlwdWxhdG9ySUQiOiIiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiRGltZW5zaW9ucyI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9O3siQVBNTERWIjp7IngiOjcuNzU2NjY1MjI5Nzk3MzYzLCJ5Ijo0LjI0NjA3NzUzNzUzNjYyMSwieiI6MS40MzU5OTc0ODYxMTQ1MDJ9LCJBbmdsZXMiOnsieCI6LTQ1LjAsInkiOjYwLjAsInoiOi03NS4wfSwiQXRsYXNTcGFjZU5hbWUiOiJhbGxlbl9tb3VzZV8yNXVtIiwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoiIiwiU2VsZWN0aW9uTGF5ZXJOYW1lIjoiZGVmYXVsdCIsIlR5cGUiOjI0LCJDb2xvciI6eyJyIjowLjA5MDE5NjA4MDUwNTg0NzkzLCJnIjowLjc0NTA5ODA1NDQwOTAyNzEsImIiOjAuODExNzY0NzE3MTAyMDUwOCwiYSI6MS4wfSwiVVVJRCI6ImY1Yzk4MmQ5LWMyM2YtNDY2Mi04NWUzLWE0ZWM3ZDVkMjcyNCIsIk5hbWUiOiJTQ20tZjVjOTgyZDkiLCJBUElUYXJnZXQiOiIiLCJOdW1BeGVzIjowLCJNYW5pcHVsYXRvcklEIjoiIiwiWmVyb0Nvb3JkT2Zmc2V0Ijp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjAsInciOjAuMH0sIkRpbWVuc2lvbnMiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkJyYWluU3VyZmFjZU9mZnNldCI6MC4wLCJEcm9wMlN1cmZhY2VXaXRoRGVwdGgiOnRydWUsIklzUmlnaHRIYW5kZWQiOmZhbHNlfTt7IkFQTUxEViI6eyJ4Ijo0LjkzOTE1NjUzMjI4NzU5OCwieSI6My45ODkwNDUzODE1NDYwMjA3LCJ6IjoxLjkyMTUyMDIzMzE1NDI5Njl9LCJBbmdsZXMiOnsieCI6LTEzNS4wLCJ5Ijo2MC4wLCJ6IjotNzUuMH0sIkF0bGFzU3BhY2VOYW1lIjoiYWxsZW5fbW91c2VfMjV1bSIsIkF0bGFzVHJhbnNmb3JtTmFtZSI6IiIsIlNlbGVjdGlvbkxheWVyTmFtZSI6ImRlZmF1bHQiLCJUeXBlIjoyNCwiQ29sb3IiOnsiciI6MC41NDkwMTk2MzQ3MjM2NjMzLCJnIjowLjMzNzI1NDkxMTY2MTE0ODA5LCJiIjowLjI5NDExNzY1OTMzMDM2ODA2LCJhIjoxLjB9LCJVVUlEIjoiNWM5MTBhZGYtZmFlMC00ZGNmLTk0ZmMtMDc0ZDhkNjNhNjY4IiwiTmFtZSI6IjVjOTEwYWRmIiwiQVBJVGFyZ2V0IjoiIiwiTnVtQXhlcyI6MCwiTWFuaXB1bGF0b3JJRCI6IiIsIlplcm9Db29yZE9mZnNldCI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wLCJ3IjowLjB9LCJEaW1lbnNpb25zIjp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjB9LCJCcmFpblN1cmZhY2VPZmZzZXQiOjAuMCwiRHJvcDJTdXJmYWNlV2l0aERlcHRoIjp0cnVlLCJJc1JpZ2h0SGFuZGVkIjpmYWxzZX07eyJBUE1MRFYiOnsieCI6Mi44NjYwMjQ5NzEwMDgzMDEsInkiOjMuOTk5OTk5NzYxNTgxNDIxLCJ6IjoyLjk2NDEwMTU1Mjk2MzI1N30sIkFuZ2xlcyI6eyJ4IjoxODAuMCwieSI6NjAuMCwieiI6LTkwLjB9LCJBdGxhc1NwYWNlTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0iLCJBdGxhc1RyYW5zZm9ybU5hbWUiOiIiLCJTZWxlY3Rpb25MYXllck5hbWUiOiJkZWZhdWx0IiwiVHlwZSI6MjQsIkNvbG9yIjp7InIiOjEuMCwiZyI6MC41OTYwNzg0NTU0NDgxNTA2LCJiIjowLjU4ODIzNTMxODY2MDczNjEsImEiOjEuMH0sIlVVSUQiOiIwZmQ0ZTg3ZS1lODYzLTQwYzUtOTNhMC00ZTQ0YWQ0MjZiNTkiLCJOYW1lIjoiTU9zLTBmZDRlODdlIiwiQVBJVGFyZ2V0IjoiIiwiTnVtQXhlcyI6MCwiTWFuaXB1bGF0b3JJRCI6IiIsIlplcm9Db29yZE9mZnNldCI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wLCJ3IjowLjB9LCJEaW1lbnNpb25zIjp7IngiOjAuMCwieSI6MC4wLCJ6IjowLjB9LCJCcmFpblN1cmZhY2VPZmZzZXQiOjAuMCwiRHJvcDJTdXJmYWNlV2l0aERlcHRoIjp0cnVlLCJJc1JpZ2h0SGFuZGVkIjpmYWxzZX07eyJBUE1MRFYiOnsieCI6Ni43NDExODA4OTY3NTkwMzMsInkiOjMuOTkzNjExMzM1NzU0Mzk0NywieiI6MS4xNTEwMTE3MDUzOTg1NTk2fSwiQW5nbGVzIjp7IngiOi05MC4wLCJ5Ijo2MC4wLCJ6IjotNzUuMH0sIkF0bGFzU3BhY2VOYW1lIjoiYWxsZW5fbW91c2VfMjV1bSIsIkF0bGFzVHJhbnNmb3JtTmFtZSI6IiIsIlNlbGVjdGlvbkxheWVyTmFtZSI6ImRlZmF1bHQiLCJUeXBlIjoyNCwiQ29sb3IiOnsiciI6MC44NTg4MjM1Mzc4MjY1MzgxLCJnIjowLjg1ODgyMzUzNzgyNjUzODEsImIiOjAuNTUyOTQxMjAzMTE3MzcwNiwiYSI6MS4wfSwiVVVJRCI6ImY0YzZjOGZiLWZlNzctNDc1Ny05MmYwLWFkNmRiMGJjOTM1NiIsIk5hbWUiOiJmNGM2YzhmYiIsIkFQSVRhcmdldCI6IiIsIk51bUF4ZXMiOjAsIk1hbmlwdWxhdG9ySUQiOiIiLCJaZXJvQ29vcmRPZmZzZXQiOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMCwidyI6MC4wfSwiRGltZW5zaW9ucyI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQnJhaW5TdXJmYWNlT2Zmc2V0IjowLjAsIkRyb3AyU3VyZmFjZVdpdGhEZXB0aCI6dHJ1ZSwiSXNSaWdodEhhbmRlZCI6ZmFsc2V9&Settings=eyJTaG93U3VyZmFjZUNvb3JkIjp0cnVlLCJTaG93QnJlZ21hQXhpcyI6dHJ1ZSwiU2hvd0luUGxhbmVTbGljZSI6dHJ1ZSwiR2hvc3RJbmFjdGl2ZVByb2JlcyI6dHJ1ZSwiR2hvc3RJbmFjdGl2ZUFyZWFzIjpmYWxzZSwiVW5pdHNJblVNIjp0cnVlLCJTaG93QWxsUHJvYmVQYW5lbHMiOmZhbHNlLCJQcm9iZVBhbmVsSGVpZ2h0IjoxNDQwLjAsIkRldGVjdENvbGxpc2lvbnMiOnRydWUsIlJvdGF0ZUFQTUwyUHJvYmVBeGlzIjp0cnVlLCJQcm9iZVNwZWVkIjozLCJTaG93QXRsYXMzRFNsaWNlcyI6MCwiUmVsYXRpdmVDb29yZCI6eyJ4IjowLjAsInkiOjAuMCwieiI6MC4wfSwiQXRsYXNUcmFuc2Zvcm1OYW1lIjoibnVsbCIsIkJyZWdtYUxhbWJkYVJhdGlvIjoxLjAsIkVwaHlzTGlua1NlcnZlcklQIjoiIiwiRXBoeXNMaW5rU2VydmVyUG9ydCI6ODA4MSwiRXBoeXNMaW5rUmlnaHRIYW5kZWRNYW5pcHVsYXRvcnMiOiIiLCJBY2NvdW50c0xvZ2luVG9nZ2xlIjp0cnVlLCJPcGVuRXBoeXNBUElUb2dnbGUiOmZhbHNlLCJPcGVuRXBoeXNBUElUYXJnZXQiOiJodHRwOi8vbG9jYWxob3N0OjM3NDk3IiwiU3Bpa2VHTFhBUElUb2dnbGUiOmZhbHNlLCJTcGlrZUdMWEFQSVRhcmdldCI6IjEyNy4wLjAuMTo0MTQyIiwiU3Bpa2VHTFhIZWxsb1BhdGgiOiJDOlxcSGVsbG9TR0xYLXdpblxcSGVsbG9TR0xYLmV4ZSIsIkFQSVVwZGF0ZVJhdGUiOjEwLjAsIlVzZUlCTEFuZ2xlcyI6ZmFsc2UsIkF4aXNDb250cm9sIjp0cnVlLCJVc2VBY3JvbnltcyI6dHJ1ZSwiVXNlQmVyeWwiOnRydWUsIkNhbWVyYVpvb20iOjUuMCwiQ2FtZXJhUm90YXRpb24iOnsieCI6MC4wLCJ5IjowLjAsInoiOjAuMH0sIkF0bGFzTmFtZSI6ImFsbGVuX21vdXNlXzI1dW0ifQ==) about two weeks ago, at which time it produced this layout: 

[image]

but now upon clicking it, I think you will see this, where they are off to the side and I think not as deep. The selected brain regions are also deselected, if it's intended that those are part of the link. I can't tell from me end whether this is a bug with the links or whether there was a change recently that broke link compatibility, but I assume it is one of those two thing?  

[image]

dbirman, 2024-11-06:
Definitely a bug that's changing something. Thanks for flagging this!

kjy5, 2024-11-11:
Regions getting deselected is probably a bug, but the location of the probes seems right. Referencing what I said in #752 it looks like the reference coordinate is back at Bregma in the second picture whereas in the first picture, it was off somewhere else.

It's still probably a bug that this isn't loaded correctly, but the probes are in the right relative positions (I think).

dbirman, 2024-11-11:
The bug is that reference coordinate information isn't being saved and/or loaded properly, and I think the defaults are also messed up somehow. All of this should be fixed by the SceneData changes, so I probably won't patch this before getting that rolled out.

kjy5, 2025-04-11:
Will be fixed with new SceneData system

closed 2025-04-11 by kjy5