VirtualBrainLab/Pinpoint #755: is that possible to change the neuropixiel probe to surgery forceps

zclecle2, 2024-12-02:
**Is your feature request related to a problem? Please describe.**
It is hard to introduce my surgery to others without a good 3d video demo

**Describe the solution you'd like**
is that possible to change the neuropixiel probe to surgery forceps

**Describe alternatives you've considered**
many 3d mouse brain atlas viewer can only be viewed interactively, but does not have a external object manipulation function

**Additional context**
[image]

kjy5, 2025-04-09:
Could you send us a 3D model like a to-scale OBJ file of your forceps? We can use that in Pinpoint.

zclecle2, 2025-04-10:
Sorry, I do not have that.

kjy5, 2025-04-10:
Ok, it’s a lower priority item for us at the moment but we can get something in. In the meantime I suggest using a NP 1.0 probe or the pipette as a substitute.