VirtualBrainLab/Pinpoint #758: Issues with offline functionality usage

yejindeus, 2024-12-17:
Hello author, while using Pinpoint, I encountered a situation where it needs to be used in an offline environment. When I was browsing the BrainAtlas source code, I found that the localAddressables in "public boolean localAddressables=false;" were not called. Is this boolean value used to enable offline loading entry? I also saw a user asking if there is an offline function in the issue before browsing. You said you would add it later, but I don't know if it has been added?

dbirman, 2025-01-02:
@yejindeus Hi thanks for checking in about this. Pinpoint on windows and linux can be run offline by launching the app once with an internet connection to allow the atlas cache to be downloaded. After that initial run the app should work offline.

There is no plan to add a fully offline mode in the future, our primary focus is on features for the browser web app.

closed 2025-01-02 by dbirman

yejindeus, 2025-01-03:
> @yejindeus Hi thanks for checking in about this. Pinpoint on windows and linux can be run offline by launching the app once with an internet connection to allow the atlas cache to be downloaded. After that initial run the app should work offline.嗨，感谢您的查看。Pinpoint 在 Windows 和 Linux 上可以离线运行，只需使用 Internet 连接启动应用程序一次，即可下载图集缓存。在初始运行之后，应用程序应该可以离线工作。
> 
> There is no plan to add a fully offline mode in the future, our primary focus is on features for the browser web app.将来没有添加完全离线模式的计划，我们的主要重点是浏览器 Web 应用程序的功能。

Thank you for your reply, author. While waiting for your reply, I have written a method: store the complete information cached for the first time in the StreamAssets folder directory. When the current network is disconnected, the cached information in this directory will be copied to the corresponding directory on the C drive. This way, as long as the cached files are correct, they can be used offline directly. Thank you again, author, and your team for your open-source work, which has been very helpful to me.