VirtualBrainLab/Pinpoint #767: Standalone build 1.1.7 from Steam does not show electrode location

vncntprvst, 2025-03-28:
**Describe the bug**
I installed the standalone build on a Windows machine and the probe UI is empty (does not show the electrodes location). The in-plane slice is also empty. This is not the case in the web GUI. 

**To Reproduce**
Install latest version from Steam (as of 03/27/25).
Add a probe to the scene and lower it to any coordinates.
The issue is not probe specific.

**Screenshots**

[image]

[image]

**Log**

[Player-032825.log](https://github.com/user-attachments/files/19506740/Player-032825.log)

dbirman, 2025-03-28:
Thanks for the heads up, I will release a new version

kjy5, 2025-04-11:
v2.0.0 of Pinpoint has been released on Steam!

closed 2025-04-11 by kjy5