VirtualBrainLab/Pinpoint #768: Coordinates with Manipulator and Pinpoint software

20180220, 2025-03-31:
Hello,  

We are using Pinpoint software (Pinpoint v1.1.7). Also we tried the one from Steam. 
Our seup is  Pinpoint software (Pinpoint v1.1.7) with a uMp-3-NP Manipulator of Multiprobe rig (Sensapex).
We are wondering if there is any way to match the coordinates in the software 

[image]

kjy5, 2025-04-09:
Hi,
Support for the uMp-3 is not quite ready yet. There was an early version of it in Pinpoint v1 but that was for testing purposes only. We are working on implementing uMp-3 support in v2 of Pinpoint along with the multiprobe rig.

DV is strictly the vertical axis while the depth axis would be along the manipulator's x-axis. If your probe is at an angle, DV and depth would not add up to be the x-axis.

Again, v1 of Pinpoint does not properly support the uMp-3 and we are working on fixing this for v2. Please consider switching to v2 of Pinpoint!

closed 2025-04-09 by kjy5

kjy5, 2025-04-10:
Please see #769 for updates on this!