VirtualBrainLab/Pinpoint #769: Keyboard use for Pinpoint Software

20180220, 2025-03-31:
Hello,

We believe that Pinpoint can be controlled by the keyboard as shown in the picture below,
but somehow we are unable to do so. Our set up is Pinpoint software (Pinpoint v1.1.7) with a uMp-3-NP Manipulator of Multiprobe rig (Sensapex).
[image]
Is there something we sould do to activate the keyboard?
We apprecaite your support.

Best Regards,

kjy5, 2025-04-09:
Following my comment on #768 the uMp-3 manipulator are not completely supported in v1 of Pinpoint. To manually control a manipulator using your keyboard, please refer to the Ephys Link connection instructions to link manipulators to Pinpoint: https://virtualbrainlab.org/pinpoint/tutorials/tutorial_ephys_link.html#mirror-the-positions-of-in-vivo-probes-in-pinpoint. Specifically, in section 2, the 8'th marker points to a check box you can use to enable manual control. Note that since probes are locked by default you will need to use the <kbd>Space</kbd> bar to unlock the probe and send keyboard commands.

20180220, 2025-04-10:
Thank you very much for supporting us!
Now we know that the uMp-3/multiprobe rig is not supported completely in v1 of Pinpoint. 
We checked the box for 8th marker point and used space bard to unlock but the key board commands did not  become valid/effective.  
Maybe becuase it is because v1? Should we try it in v2 of Pinpoint? 

Best Regards,

20180220, 2025-04-17:
Thank you very much for your support. Yes, then we will wait for a versin that will be able to support uMp-3 with Multiproble rig. 

Best Regards,

20180220, 2025-04-17:
Wow! yes it is. We will try and let you know! THank you  very much!

20180220, 2025-04-23:
Hello,

We tried it but it did not work well. 
We cannot move the manipulator from the PC. 
Only Return Zero Coordinate button works from the PC. 
Is the software working there? 
Is it possible  for you to help us online meeting? 

Best Regards,

kjy5, 2025-04-23:
Can you confirm if the position is being correctly displayed? My understanding is that you've attached the manipulator and you can properly see its movements when you move it from the wheel controllers. Then you enabled manual control and tried using the keyboard to move and nothing happened. Check to see if the probe is locked (numbers are grayed out on the right). Use the space bar to unlock it. If that didn't work use the shift key to increase the input speed. 

Let me know if those two steps helped or if you're getting stuck elsewhere!

20180220, 2025-04-24:
Yes, that is correct. That was the situatin, but we tried the space bar to unlock and we were able to use the PC to control manipulator.
Thank you! but we have a few addtional questions. 

1. The numbers displayed on the TSC and PC do not match. Is it possible for us to match? 
2. When the numbers displayed on X and Z increase, the number displayed on DV and AP decrease. Could you fix it? or are we doing something wrong? 
3. We checked "Enable Copilot(Beta)" and "Enable New UI(Experimental)" but it seems that any functions there do not work. The functions are not available? 

We appreciate your support.

Best Regards,

kjy5, 2025-04-24:
That's great you have them connected now! To answer your questions:

1. You need to calibrate the virtual probe to your real one. With your manipulators, move them to Bregma. Then in Pinpoint, press the home icon on the right panel for the corresponding probe to reset its virtual position to Bregma as well. Now they are in sync. The numbers still won't match though because the Sensapex controllers are showing you the translation stage values and Pinpoint is showing you CCF coordinates. But they will change in sync.
2. That is totally normal. X and Z are not necessarily related to the AP, ML, DV coordinates your probe is moving in depending on your probe's angles. AP/ML/DV are the CCF coordinate and your X and Z coordinates are your manipulator's raw translation values.
3. Copilot is the automation system that will move the probes for you in experiments. Because it's an experimental feature we would love to set up a meeting with you to chat about it and work with you to get set up using it! The internal tutorial we use for our researchers can be found here: https://virtualbrainlab.org/pinpoint/tutorials/tutorial_ephys_copilot.html (you should skip part 2 and instead create probes you intend to use for planning and separate dummy probes for visualization, which you have already been doing here)

Let me know if you have time to set up a video call meeting to chat about using the automation system! If you can get this to work with just the tutorial you are welcome to simply carry on.

20180220, 2025-04-25:
Thank you very much for your explanation! NOw we know better about Pinpoint software. 

About no.3, yes, we would like you to help us use teh Cipilot feature. How could we set up a meeting with? We are in Japan. Hopefully, we can find a good timing between us. 

Actually, we checked the trutorial but it seems somehow the version of Pinpoint does not have "account" on the left in the category that appears in the tutorial so we would need your help with that as well.. 

Anyway, we appreciate your support.

kjy5, 2025-04-25:
Yes, we can set up a meeting. Do you use Zoom meetings? What is a good email for me to send you a meeting?

I am in the Pacific time zone. I can usually meet on Tuesdays or Wednesdays from 4 PM Pacific time onwards (8 AM onwards for you). Please let me know what days work for you! I am busy on 4/30 at around these times, but I am otherwise available.

20180220, 2025-04-30:
Yes, we do, could you send the meeting to info@intermedical.co.jp? 
For instance, would it be possible for us to have meeting on 5/6 at 6pm in your time/5/7 at 10am in our time? 
We are sorry for making you work late, but if we could meet around the time, it would be really appreciated. 
Again, thank you very much for your suppot.

Best Regards,

kjy5, 2025-04-30:
That time will work for me! I've sent you an invite.

20180220, 2025-04-30:
Really? it seems that I have not been able to receive it yet... I think i did not make a mistake with the address... or I cannot find it in junk mail either... Could you try to send it again?

kjy5, 2025-04-30:
I sent the meeting through Zoom, but I just sent it again via email. Let me know if you find it this time!

20180220, 2025-05-01:
Yes,  I got it. Thank you!